prusa3d/PrusaSlicer #1706: 1.42.0alpha3: soluble interface is malformed

supermerill, 2019-01-24:
### Version
win64 1.42.0alpha3

### Behavior
using the support with soluble interface parameter, with 2 interface layers, on the 3 interface layers dawned, the bottom2 are ok (good simple rectangle) but the top one (the most important) has some " notches that totally deform it.

[image]

#### STL/Config (.ZIP) where problem occurs
[support_notch_problem.zip](https://github.com/slic3r/Slic3r/files/2792758/support_notch_problem.zip)

bubnikv, 2019-01-26:
That is the "interface loops" settings.

bubnikv, 2019-01-29:
Does it answer your issue? Can we close this issue?

bubnikv, 2019-01-29:
> Sorry, i'm not at home right now. I don't understand why "interface loop"
> can't create quare perimeter for the interface layers. why these notches?
> Or if you know, where something in supportmaterial.cpp is doing that?

class LoopInterfaceProcessor

We never use it, I just re-implemented what was there in Perl with the hope
that it may be useful to somebody. I suppose the idea was to have at least
some lines at the interface always perpendicular to the perimeters,
therefore supporting it. With the rectilinear supports you always risk,
that the perimeter will fall through the gap between two parallel lines. I
suppose the loops were designed to mitigate that risk.

supermerill, 2019-01-29:
Thanks, i will look at it.

On my fork, i changed the interface pattern from concentric to rectilinear for soluble support, waiting for a fix.

You can close the issue if you want.

bubnikv, 2019-01-29:
The concentric pattern is there on purpose, as the rectilinear impresses nasty waves to the sides of an object.

Closing

closed 2019-01-29 by bubnikv