prusa3d/PrusaSlicer #1707: Feature request: Organisation managed profiles

jokohoko, 2019-01-24:
### Version
1.41.2

### Operating system type + version
All systems

### Behavior
I organize a bunch of printers at our University of Antwerp (Belgium). And would like to make some standard settings choices simpler for the students and colleagues. 
What i would love to have would be like the additional boards manager url's in Arduino. Where settings can be managed by the organisation. So that when students enter the url in there, they wil get the possible profiles and materials available at our printing room. Without mistakes.

Like you guys host the profiles update on your site that gets checked and updated if anything gets changed it would be great to provide such custom url that organisations can manage and update remotely. This might also help out a bunch of Fablab's and such.

This is a feature request

#### STL/Config (.ZIP) where problem occurs

supermerill, 2019-01-25:
I think you can already do that.

Did you try to copy & rename the prusaresearch.ini in slic3r/resources/profiles?
I've done that for my father printer, so it can "install" it via the menu.

I didn't tested the config_update_url, but i think it will also work as intended.

bergos, 2021-10-31:
It is not possible to create a custom profile with a custom `config_update_url` property for automatic updates. [These lines](https://github.com/prusa3d/PrusaSlicer/blob/dac810951daeec3283cecd1e9bbe5b17a3998a5b/src/slic3r/Utils/PresetUpdater.cpp#L357-L363) in the code prevent loading updates from any URL different to `http(s)://files.prusa3d.com/wp-content/uploads/repository/`.

If @prusa3d really thinks it's too dangerous to load profiles from other hosts out-of-the-box, please give us an option to overwrite this check in the settings.

closed 2026-09-29 by prusaslicer-bot[bot]