prusa3d/PrusaSlicer #1713: Cut feature closing itself / gizmo freezes

th33xitus, 2019-01-26:
### Version
1.42.0-alpha3+win64

### Operating system type + version
Win 7 Ultimate

### Behavior
Cut function aborts when using a multifunction key on keyboard.
I wanted to quickly calculate what height i need to cut the bottom of an object, so i pressed my calculator key on the keyboard, calculator opened and as soon as i moved my mouse cursor the cut-gui disappeard and the gizmo froze.
Same happens when using the browser key for opening a new browser tab or any other kind of key which opens a program and brings up its window into the foreground.

[image]

As you can see on the screenshot, gui disappeard, gizmo still there but not able to move it. Need to enter the cut function again to be able to move it again. Sometimes the gizmo doesn't freeze, but still the gui disappears which make reopening it necessary

sgano, 2019-01-26:
I see similar behavior on MacOS (on v1.42.0 alpha 3) when the cut dialog is up and I hit the space bar (the gizmo stays up but the dialog disappears).  Also I am unable to type into the cut dialog to specify an exact hight to cut the model on MacOS (which I figured is related to this issue).

vojtechkral, 2019-01-28:
Ok, I can reproduce the bug. I'll have a look.

FWIW the SLA support points gizmo seems to have the same problem.

enricoturri1966, 2019-01-29:
Working on it -> http://dev.prusa3d.com/browse/SPE-801

FidelCapo, 2019-02-07:
This is issue should be fixed in version 1.42.0-alpha5. Can you confirm it? Thank you.

sgano, 2019-02-09:
I can confirm this issue is fixed in 1.42.0-alpha5 on MacOS.