prusa3d/PrusaSlicer #1716: 1.42.0-alpha3 unable to configure soluble support material w/ wipe tower UNLESS starting from a MK3 Soluble* profile

jhoblitt, 2019-01-26:
### Version

`Slic3rPE-1.42.0-alpha3+linux64-full-201901111425.AppImage`

### Operating system type + version

```
$ cat /etc/redhat-release 
Fedora release 29 (Twenty Nine)
```

### Behavior

* Attempting to enable the wipe tower when a soluble support material extruder is configured generates two pop-up dialog errors that claim to only apply to non-soluble support material when a regular MK3 print profile is selected.  Oddly, selecting one of the preset MK3 Soluble* profiles, then switching back to a non-soluble print profile seems to prevent the error dialogs from appearing.
* Even if this is supposed to be an error condition (why?), there should not be two pop-up error dialogs from the same problem and the wording of the error messages is, at best, confusing.
* Note that the UI rendering is either broken or not completely grayed out when the error dialogs are visible.

[image]
[image]
[image]
[image]
[image]

#### STL/Config (.ZIP) where problem occurs

[cat_ball-v2.3mf.zip](https://github.com/prusa3d/Slic3r/files/2799386/cat_ball-v2.3mf.zip)

FidelCapo, 2019-01-29:
I think this setting can be configure when you first set Print settings - Support material Contact Z distance to 0 (soluble) and then set extruders and wipe tower. Did it solve your issue?

bubnikv, 2019-04-13:
no response, closing

closed 2019-04-13 by bubnikv