prusa3d/PrusaSlicer #1717: Cannot assign a color\filament\extruder to a MMU object

RHTizzy, 2019-01-26:
### Version
1.42.0-alpha3+win64

### Operating system type + version
Windows10 64bit

### Behavior
* _Describe the problem_
On MMU2 print it is not possible to assign a color/filament/extruder to object parts

* _Steps needed to reproduce the problem_
In previous versions the settings pop-up allowed user to assign an extruder per part. The pop-up for settings is no longer available. In expert mode the user can add settings per part for extruders but every settings gives an obscure error about non-soluble supports. Simple or Advanced mode have no options at all for assigning an extruder to a part

#### STL/Config (.ZIP) where problem occurs

lukasmatena, 2019-01-27:
I cannot reproduce the issue, changing extruders for object parts works correctly for me. Could you please expand about the obscure error and ideally share your 3MF with configuration? I think that is more likely to be the actual problem.

RHTizzy, 2019-01-28:
Sorry, already deleted the Alpha as it was not usable without being able to change the appointed extruder. It shouldn't be too difficult to find the error in the code if you're really interested. 

I must add that the process of right clicking to add a setting per part and then having to appoint the extruder is quite tedious for such a basic function. Also adding a setting was not available for basic mode which also makes no sense as even basic users will want to print multimaterial. Right clicking should give the option directly to assign a different filament.

In extension of this issue, it's also unclear how to assign a soluble filament as support material. These things are quite basic IMHO and should have easy access for users.

bubnikv, 2019-01-28:
Cannot you open the extruder drop down box?

[image]

RHTizzy, 2019-01-28:
That section was not on my screen. I will reinstall to verify and get back to you.

RHTizzy, 2019-01-28:
Issue can be closed. It needs a double click to make the extruder selector dropdown visible, which I had not tried. I also found the setting to assign supports to an xtruder for solubles and such which was in the printsettings\multiple extruders section. I add it here in case someone else has the same confusions. Apologies for wasting your time and keep up the good work.

bubnikv, 2019-01-29:
> It needs a double click to make the extruder selector dropdown visible

It is an unfortunate feature of the wxWidgets UI framework

Closing

closed 2019-01-29 by bubnikv