prusa3d/PrusaSlicer #1720: Feature request: Different layer heights per part

Anima1000000, 2019-01-27:
For some prints I would like to apply various steps of layer heights to different parts.

For instance the visible outer perimeters have a layer height of 0.1 mm while less demanding parts, like infill or inner perimeter have a layer height of 0.2 mm. When handling the gcode it should first print two fine layers of the 0.1 mm and then the 0.2 mm. Any multiplication of the finer layer height should work.

It's a similar function to "combine infill every n layers", but allows of more control.

grafzahl, 2019-01-28:
Did not give it a try yet, but for me this is working in the actual alpha3 - i can modify the Layerheight of each individual Object.

Anima1000000, 2019-01-28:
Good to hear this is working for you, I might be overlooking something.

I am running Slic3r 1.42.0-alpha3 on MacOS 10.13.6.

With only one object I have the possiblity to adjust the Layer Height in the Add Settings menu. But with multiple parts this option is unavailable, neither is it possible to load more parts when this option has been chosen.

Anima1000000, 2019-01-28:
The layer height editing tool applies the same layer height to all object / parts. The feature request is for different layer heights per object at the same z height. Request still stands.

But good to see the layer height tool got fixed.

wabbitguy, 2019-01-28:
I'm running Mac OS 10.13.6 as well. In the add settings I can add perimeters and infill and completely control each objects settings. After adjusting two objects, I loaded a 3rd item and let it at the normal default settings. It all slices perfectly.

The object top left sliced .2 layer height, 10% IF, bottom right .05, 40% IF. Washer is .3 LH.

[image]

[image]

wabbitguy, 2019-01-29:
Great to hear you got a work around, or at least figured out. I never thought of using it with the "add parts". I only use those for blocking out supports.

bubnikv, 2019-01-29:
You can certainly set layer height per object. You cannot set layer height per part, as all parts are sliced together, sharing single slicing table. As of now, you can change the layer height per vertical segment using the variable layer height editing tool. In the future, we plan to provide a way to add layer height span attributes to object settings, but it will likely not come into this release.

closed 2019-01-29 by Anima1000000

wabbitguy, 2019-01-29:
@bubnikv I just have to say the amount of work that has gone into Slicer is astounding and I for one, am extremely grateful for all of your hard work! Thank you!