prusa3d/PrusaSlicer #1723: Thin wall torture test

wavexx, 2019-01-27:
### Version
282ef552f9111583dc9bcfba641ecd41c74353ff

### Behavior
There are several issues when generating perimeters on "thin" walls. I experienced a few and commented on #1389 as well. I was able to generate a torture test model. These show up when doing mechanical parts that have the nozzle size in mind. As will be evident, "detect thin walls" is not a panacea.

The test case is designed for a 0.2 layer height, 0.4 nozzle.

The model includes 5 rows that have a common *minimum* width, with 5 columns containing 5 test cases. The rows have the following minimum widths: 0.4, 0.8, 0.86, 1.2, 1.26mm. 0.86 and 1.26 are the values suggested by slic3r for this nozzle/layer height, which I honestly never use ;)

The first 2 columns are wedges with a 1.2 and 0.4 vertical wall. The minimum width of the vertical wall is calculated as above. The third column is simply going through a minimum width. The fourth column is just a plain vertical wall. The fifth is a 0.5 rho conic going from vertical to horizontal intended to stress all orientations.

Let's go through the problems. Current slic3r on master with 0.4 nozzle, 0.2 layer height, thin walls detection off:

[image]

- In the first row all 0.4 walls are missing. There's actually a gap between the wedge of the third column.
- On the fifth row (1.26) the conic shows an alternation between perimeter an infill. I get this is probably caused by numerical instability, however this causes thin structural walls to fail. It is not exclusive to this configuration: it happens frequently at other wall sizes and different layer heights that *should* just be filled with an extra perimeter. Also, it slows down considerably the print process by travels and retractions.

Same configuration with "thin walls" detection on:

[image]

Now things get even more interesting:

[image]

The first and third wedge with the 0.4 walls are missing, although the vertical wall and the conic are both correctly generated.

The wedges show a problem which is common to all cases where small features share a "thin" wall. Slic3r favors the perimeters of the small areas and fill the remaining thin walls independently (if at all).

I consider this to be bad behavior, as it causes more retractions and travels where a single perimeter (starting and ending at the stricture) would have done and resulted in a stronger structural part in all 3 cases.

I know @supermerill has done some work in #1028 and can maybe test this model.

#### STL/Config (.ZIP) where problem occurs
[walltorture.zip](https://github.com/prusa3d/Slic3r/files/2800508/walltorture.zip)

supermerill, 2019-01-28:
you forgot to link your settings

here with my version: 
[image]

wavexx, 2019-01-28:
The settings are the stock "0.2 FAST" for the MK3 in this case.
Your results are *way* better than the current master.

Since we're here, can I point out some possible improvements?

[image]

Would it be possible to connect the perimeters of the wedges in the order as shown above? The order/direction doesn't matter much, however as I wrote before, it would be very helpful to fill the line connecting 2-3 in a single pass. When there are multiple small features connected by a shared wall, the shared wall should be preferred. I don't know how internally the perimeter generation works, but maybe favoring segments that result in longer walls would work?

If that little line gap is filled alone, it often results in a part which is not as strong and with visible artifacts on the exterior.

I also notice there's an extra cusp (outlined with the circle) in all the corner of the wedges. Is this intentional?

supermerill, 2019-01-28:
> Would it be possible to connect the perimeters of the wedges in the order as shown above? The order/direction doesn't matter much, however as I wrote before, it would be very helpful to fill the line connecting 2-3 in a single pass.

Not possible, thin walls are printed after all perimeter to ensure there are something to anchor them. Changing the logic for this very corner case isn't worth the hassle for my free time.
Internally, there are the perimeters in an array, stored as loop and there are an other array of thin walls.

> If that little line gap is filled alone, it often results in a part which is not as strong and with visible artifacts on the exterior.

I anchored them more deeply, hoping it resolve the issue.

> I also notice there's an extra cusp (outlined with the circle) in all the corner of the wedges. Is this intentional?

yes, there are a tiny pointy bit that could be extruded as thin wall. The tip is larger than the reality because we are limited by the extruder nozzle size. In my proposal, i added recently a new parameter that let us say "what is the min width to takes into account". If this is smaller than the nozzle (default to 33% of the nozzle size), the very thin areas will be over-extruded a bit.

Here, the algorithm say "hey, the perimeter have let a big enough area to be filled by a thin wall!" and he succeed to create something big enough to pass the filters.

mdealer, 2019-03-30:
I also think it's sort of a bug. Nearly every model I print contains some area where Slic3r (sorry for exaggerations, but I'm done hoping for the best) craps out and does 20+ retractions and unnecessary moves to fill some uninteresting gaps. This causes multiple tiny extrusions with huge retractions inbetween, needless to say this causes weird behavior on some configurations with low extruder steps per mm.

This part needs some major refactoring. Something like dynamic extrusion width to produce sharper corners etc instead of filling tiny gaps afterwards.

wavexx, 2019-03-31:
On Sat, Mar 30 2019, Edgars Batna wrote:
> This part needs some major refactoring. Something like dynamic
> extrusion width to produce sharper corners etc instead of filling tiny
> gaps afterwards.

Overall this is an area where Cura fares a bit better. The outer
perimeter becomes a single contour in this torture test. Cura is also
more keen to use variable extrusion width. But Cura is not a panacea
either: I have a long list of issues with it as well (notably, infill is
always weak).

For instance, I would personally consider filling the cusps I outlined
before a bug: they always result in sharp and uneven fills. If you're
printing a screw-like structure, or a slanted corner, for example, they
will make the edge totally unusable. This is one of the reasons you
often need to toggle "detect thin walls".

rennne, 2019-05-16:
Gap fill behaves inconsistent with the same structures at different angles

The picture shows a part of a inverse tooth wheel, all teeth are the same some are filled and others are not. The only difference between them is the angle.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]