prusa3d/PrusaSlicer #1724: Absolute-E coordinate wipe incorrect retract length

FormerLurker, 2019-01-27:
### Version
1.41.2+win64

### Operating system type + version
Windows 10 pro 64 bit

### Behavior
I sliced a file using absolute E coordinates by unchecking Printer Settings->General->Advanced->Use relative E distance, and noticed that the wipe action extruded a lot of filament right after a layer change (a sizable blob). After reviewing the gcode, I believe there is a slight bug caused by G92 E0.0 command coming before the wipe action.  Here is the gcode that was produced (annotated by me) starting on line 254:
```
; up to this point it's been extruding
G1 X128.697 Y100.870 E6.95640
G1 X129.300 Y101.472 E6.98345
M106 S255
;BEFORE_LAYER_CHANGE
G92 E0.0 ; This command resets the E coordinate to 0, and should not be included
;0.4

G1 F8640 ; We're starting to wipe
G1 X128.697 Y100.870 E6.78672 ; This extrudes a lot of of filament!!!!
; the previous command causes my extruder to skip a lot
G1 X128.158 Y100.870 E6.66208;  retract a bit
G1 X129.130 Y101.842 E6.34448
G1 X129.130 Y102.367 E6.22345
G1 E6.18345 F2100.00000 
; now we are retracted about  0.6mm (6.78672 - 6.18345)
G92 E0
G1 Z0.800 F10800.000 ;
;AFTER_LAYER_CHANGE
;0.4
G1 X132.567 Y98.561
G1 Z0.400
G1 E0.80000 F2100.00000 ; This will extrude an additional 0.2mm!
```

If the first G92 E0.0 were removed (there is already one after the wipe), this code would function as expected.

There are many more examples of this behavior within this file.  I just chose the example I first discovered.

#### STL/Config (.ZIP) where problem occurs
[calibration_tower_and_config.zip](https://github.com/prusa3d/Slic3r/files/2800926/calibration_tower_and_config.zip)

You'll notice that use_relative_e_distances = 0 is the only major change from the defaults.

Normally I would prefer relative E distances, but I noticed this issue after running some tests for software I'm developing.

This is the source of the tower, which I feel it's important to add: [This calibration tower](https://www.thingiverse.com/download:17276) on thingiverse by [coasterman](https://www.thingiverse.com/coasterman.
)

bubnikv, 2019-01-29:
This issue sounds familiar to https://github.com/prusa3d/Slic3r/issues/1631
Would you please retest with https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha4 ? 
Thanks

FormerLurker, 2019-01-29:
This must be a different issue.  See this output sliced with the same file using the default settings (except for absolute E extrusion) in the alpha you linked to:

```
G1 X128.158 Y100.870 E6.93927
G1 X128.697 Y100.870 E6.95640
G1 X129.300 Y101.472 E6.98345
M106 S255
;BEFORE_LAYER_CHANGE
G92 E0.0  ; this is the problem.
;0.4

G1 F8640
G1 X128.697 Y100.870 E6.78672 ; Extrudes 6.7mm!  The extrusion is too fast for the printer and it skips
G1 X128.158 Y100.870 E6.66208
G1 X129.130 Y101.842 E6.34448
G1 X129.130 Y102.367 E6.22345
G1 E6.18345 F2100.00000
G92 E0 ; This G92 is correct
G1 Z0.800 F10800.000
;AFTER_LAYER_CHANGE
;0.4
G1 X132.567 Y98.561
G1 Z0.400
G1 E0.80000 F2100.00000 ; This is also extruding too much because the printer is not retracted fully
```

Also, the new version is looking really good!  I was thrown by the 'Advanced' and 'Expert' switch for a bit, but eventually figured it out ;)

FormerLurker, 2019-02-05:
Just checking back in.  This issue is marked 'waiting for response' but I've provided a response already :)  Just making sure this doesn't fall through the cracks.

rtyr, 2020-06-18:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/4379 and https://github.com/prusa3d/PrusaSlicer/issues/4327. 

This issue is caused by incorrect printer profile configuration. That "G92 E0" command is custom g-code used in our printer profiles in combination with relative positioning. You shouldn't use this custom g-code if you want to use absolute positioning.

[image] 

Closing. You can continue discussion here https://github.com/prusa3d/PrusaSlicer/issues/4379 if needed.

closed 2020-06-18 by rtyr