prusa3d/PrusaSlicer #1725: "+" for color change shown in single extruder printer and not dual extruder printer

thedevleon, 2019-01-28:
### Version
Version 1.42.0-alpha3+win64

### Operating system type + version
Windows 10, 64-Bit

### Behavior
I have two printer profiles, one with 1 extruder and one with 2 extruders.
When selecting the single extruder profile and going intro "Preview", it shows a little "+" next to the layer slider to add a color change, which doesn't make sense, since there is only 1 extruder.
When selecting my second printer profile with two extruders, I can select 2 filaments at the top, but the "+" in the layer slider disappears. It seems like the logic here is inverted.

[image]

[image]

thedevleon, 2019-01-28:
I'm not 100% sure, my C++ is a bit rusty, but could this be the issue?
https://github.com/prusa3d/Slic3r/blob/master/src/slic3r/GUI/GUI_Preview.cpp#L632

`        if (config.opt<ConfigOptionFloats>("nozzle_diameter")->values.size() > 1) 
            color_print_enable = false;`

Shouldn't this be:

`        if (config.opt<ConfigOptionFloats>("nozzle_diameter")->values.size() > 1) 
            color_print_enable = true;`

lukasmatena, 2019-01-28:
> Is there a reason why color change would not be possible with >1 extruder system?

It is possible, but not currently implemented. It is not as simple as just enabling the plus button. The logic would have to branch and consider whether it is single extruder, single extruder MM or multiextruder MM printer with very different behaviour in all these cases.

As of today, the colorprint is more or less a front-end to M600 command, which does not make much sense with MM printers. The behaviour that you probably want (forcing a toolchange in certain height) can be achieved with slab modifier mesh.

thedevleon, 2019-01-28:
I see, that makes sense. Thanks for the quick reply!
I'll give modifier meshes a try then.

closed 2019-01-28 by thedevleon