prusa3d/PrusaSlicer #1727: honeycomb volumetric flowrate.

Anima1000000, 2019-01-28:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2019-01-29:
It is printed with the bridging flow to compensate for the partial overlap of extrusions in successive layers. This has been done on purpose, but I agree that it over-extrudes for thin layers, and the logic should be improved. IMHO this pattern has been superseded with the gyroid pattern, you are advised to use that instead.

Anima1000000, 2019-02-01:
I really like the gyroid infill, it had a similar issue but it was solved. However for some projects I use the 3D honeycomb as it visually fits better. It would be nice if it could be fixed, but it's not urgent since I can work around it.

closed 2026-09-29 by prusaslicer-bot[bot]