prusa3d/PrusaSlicer #1730: Multi-part object on 1.42.0 alpha 4

ACDawg, 2019-01-28:
### Version
1.42.0 alpha 4

### Operating system type + version
Windows 10

### Behavior
When loading multi-part object and selecting "Yes," objects are not loaded correctly and do not line up.

[image]
[image]

rainersc, 2019-01-29:
i can confirm this, had the same problem and went back to alpha 3

FidelCapo, 2019-01-29:
Thank you for report. Can you please send me STL file? Thank you

ACDawg, 2019-01-29:
https://www.thingiverse.com/thing:2985548

cosmith71, 2019-01-29:
Can confirm as well.  Alpha 3 works fine in this regard.

enricoturri1966, 2019-01-29:
Working on it -> http://dev.prusa3d.com/browse/SPE-799

FidelCapo, 2019-01-29:
This should be fixed in next release. Closing

closed 2019-01-29 by FidelCapo