prusa3d/PrusaSlicer #1732: output_filename_format Error

athornforeveryh, 2019-01-29:
### Version
Version 1.42.0-alpha+win64

### Operating system type + version
Windows 10 64 Bit Version 1703 Build 15063.608 

### Behavior
I'm using the default settings in Slic3r for my Mk3 and this is the first time I've ever seen this error when trying to export the G-Code:

Failed processing of the output_filename_format template.
Parsing error at line 1: Not a variable name
{input_filename_base}_{layer_height}mm_{filament_type[0]}_{printer_model}.gcode
^

If I remove the {input_filename_base}, it works. But why is it with this one file only? The file name is Loyet Print.stl. Removing the space in the name also throws the same error.

[LoyetPrint.zip](https://github.com/prusa3d/Slic3r/files/2805937/LoyetPrint.zip)

bubnikv, 2019-01-29:
This is interesting, it works for me on Windows64 even with your config file.
How can I reproduce your issue?

bubnikv, 2019-01-29:
> Hmmm... The client is only wanting the top portion so if you rotate the X
> 90, preform a cut at ~288, then rotate the x back 270, that's when I get
> the error.
>
> I went back into his original file to remove the portion I didn't want and
> exported that back into Slic3r with no issues. Just odd that rotating and
> cutting a piece would create a filename issue.

I wonder whether the cut function did not corrupt the process memory. What
if you cut, then save 3mf, then close Slic3r, restart Slic3r, reload the
3mf and reslice / export? Will you get the error? If no, then most likely
the cut function is guilty.

athornforeveryh, 2019-01-29:
Just did exactly that and no error popped up this time.

bubnikv, 2019-02-05:
Please retest with alpha6 once it is out. I have swapped the triangulation library with a proven stable one, many of the cut crashes are now fixed.

bubnikv, 2019-04-13:
no response, closing

closed 2019-04-13 by bubnikv

byteborg, 2022-10-11:
FWIW, I got this behaviour, too. It happened, when the model's y-dimension was exactly as large as the physical y-dimension of the configured printer - my reprap has a weird bed shape, just don't ask :). Making the bed shape y .1mm larger makes the problem go away.