prusa3d/PrusaSlicer #1733: incorrect coordinates label

Xuheir, 2019-01-29:
### Version
1.42.0 alpha4+full 201901281426

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
MacOs 10.14.2

### Behavior
When selecting an object in the Plater window, the coordinates shown in the Object manipulation panel for the size the Y is showing the Z value and the Z is showing the Y value.

* _Steps needed to reproduce the problem_
select an Object in the Plater window.

* _Expected Results_

* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
[image]

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2019-01-29:
The scale coordinates are those of the non-rotated object, therefore the
visual hints (the arrows) are rotated around the X axis by 90 degrees with
the object. This is intentional. If you rotated by 45 degrees instead of by
90 degress in the X axis, the Z visual hint would be tilted by 45 degrees
as well.

Most likely we will provide scaling in the world coordinates as well in the
future versions, but I don't think we will manage it in this version.

bubnikv, 2019-05-14:
the PrusaSlicer 2.0.0-rc supports rotation and scaling of object instances in the world coordinate system.

closed 2019-05-14 by bubnikv