prusa3d/PrusaSlicer #1738: Weird time estimate with travel speed

Patrik-svobodik, 2019-01-30:
### Version
1.42alpha4 with latest commits as of 30th January

### Operating system type + version
macOS Mojave 10.14.2

### 3D printer brand / version + firmware version (if known)
HEVO profile based on the MK3 profile with only the start code and firmware flavor changed, RRF2.02 stable

### Behavior
*When I change the trade speed to the reliable max of my hevo, the print time increases for some reason
*Just change the travel speed of the MK3 profile to 416mm/s

https://youtu.be/n7ZasT5Y8tQ
#### Project File (.3MF) where problem occurs
[Archiv.zip](https://github.com/prusa3d/Slic3r/files/2813954/Archiv.zip)

bubnikv, 2019-04-13:
> When I change the trade speed to the reliable max of my hevo, the print time increases for some reason

When you change the travel speed from what value to what value?

Patrik-svobodik, 2019-04-13:
As shown in the video, from the default mk3 speed to 416mm/s (max my HEVO
can handle reliably)

bubnikv, 2019-04-13:
I don't understand much, but if you are wondering why the time does not change, than it is because the movements are limited by the "machine limits" table, so your speed 416mm/s is clamped to much lower values by the firmware using the "machine limits" table.

closed 2019-04-13 by bubnikv

Patrik-svobodik, 2019-04-13:
The time changes, but it thinks it will take longer, I have set my gcode flavour to be RepRap, so the machine limits thing does not apply, did you even watch the video I linked?

bubnikv, 2019-04-13:
I did watch the video and I did not understand a thing.

Patrik-svobodik, 2019-04-13:
Essentially I slice an object with normal mk3 travel speeds, then I let it
estimate the time, then I change the speed to a higher value and reslice
it, I am expecting a slightly faster/same print time, but the print time
grows, which doesn’t make any sense

bubnikv, 2019-04-14:
I don't know whether it is correct or not. Slic3r tries to follow the logic
of the firmware, and the firmware applies a set of simple rules and
equations to ensure, that no speed or jerk limits are violated. It is quite
possible, that bumping up the travel speed 2x may actually increase the
print time slightly.

bubnikv, 2019-04-14:
This is the code to study:
https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/planner.cpp

Sebastianv650, 2019-04-14:
@bubnikv as I know the planner well, I can tell that there is no way to increase the theoretical print time by increasing a speed value. At some point it will not decrease any more as it's capped to max. speeds the printer can handle, but that's all.

@Patrik-svobodik and @bubnikv I did the test with Slic3r 1.41.1 and the latest beta 1 using identical profiles and the same model. I think it was a bug which was solved between the alpha and the beta.
I entered a bunch of travel speeds in both versions, here is the result:
[image]
[image]

As you can see, beta 1 calculates nice and logical time values whereas 1.41.1 (as your alpha) goes up into infinity after 100mm/s. Maybe some variable was rolling over?

bubnikv, 2019-04-14:
Thanks @Sebastianv650. Interestingly enough, I did a quick test today morning and I saw slight increase in print time.