prusa3d/PrusaSlicer #1739: [Request] Add [remaining_time] to custom Layer Change G-Code

dartrax, 2019-01-30:
It would be nice to have a [remaining_time] (or similar) variable available at the "Before layer change G-Code" and "After layer change G-Code" settings. This way, one can display the remaining time on the printers screen using the M117 command:

M117 Time remaining: [remaining_time]

_Is this a new feature request?_
Yes

benderl, 2019-06-11:
I had the same idea while dealing with manual color changes (no MMU) right now. I was looking for M73 codes insode the generated gcode file, but they are very sporadic. A custom gcode at layer change with M73 would help very much to estimate the time to the next filament change, so you know when to be near the printer.

As an enhanced version an "remaining time till next filament change" on the display would be perfect.

Hades32, 2020-08-09:
This would be so easy if we just had the complete time in seconds as a variable...

FYI for the time being I'm using this to at least get some progress display on my Ender 3:
```
M117 {(layer_num*100)/total_layer_count}% finished
```

closed 2026-09-29 by prusaslicer-bot[bot]