prusa3d/PrusaSlicer #1742: 1.42.0 alpha4 dragging STL file onto print bed shows "Move" rather than "Copy" or "+"

paynterf, 2019-01-31:
### Version
1.42.0 alpha4

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
MK3

### Behavior
* _Describe the problem_
Dragging an STL file from file explorer onto print bed in Slic3r works fine, but the tag on the file icon still says "Move" instead of changing to "Copy" or "+"

* _Steps needed to reproduce the problem_
Drag an STL file from file explorer window into Slic3r window

* _Expected Results_
File icon text should change to "Copy" or "+"
* _Actual Results_

[image]

paynterf, 2019-01-31:
Note that per a suggestion on the Prusa forum, I 'installed' alpha4 in C:\Users\Frank\Downloads\Slic3rPE-1.42.0-alpha4+win64-full-201901281428\Slic3rPE-1.42.0-alpha4+win64-full-201901281428 folder and run it via a shortcut pinned to my taskbar.  I suppose it is possible that the lack of a change in the icon tag is due to Slic3r not being properly installed?

Frank

bubnikv, 2019-01-31:
You may try to restart your computer, it may take the Finder some time to update.

Other than that, I have no idea how to change the "Move" to "+" or whatever else. I googled for around 15 minutes. This is something for a contributor - OSX expert - to help to solve.

bubnikv, 2019-02-01:
all right, I have marked with a "volunteer needed" flag, maybe somebody will stumble over a solution.

bubnikv, 2019-02-01:
So it was the Drag & Drop target, which was set to "Move". Now I have set it to "Copy".

commit 405d18a7 (2019-02-01): Changed default the Drag & Drop target from "Move" to "Copy". Fixes #1742
closed 2019-02-01 by bubnikv