prusa3d/PrusaSlicer #1743: Slic3r doesn't slice a mode, and crashes if I try to cancel the slicing

Liger0, 2019-01-31:
### Version
1.42.0 alpha 4

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
All printers (also custom)

### Behavior
If I load a model and I try to slice it, cura will never end the process.
If I try then to cancel it, slic3r jsut hangs and crashes.
[stuck at slicing.zip](https://github.com/prusa3d/Slic3r/files/2819218/stuck.at.slicing.zip)

#### Project File (.3MF) where problem occurs
[image]

reloxx13, 2019-02-02:
disable "avoid crossing perimeters"

Liger0, 2019-02-03:
> 
> 
> disable "avoid crossing perimeters"

Thanks, it worked.
Since most of the models just make slic3r crash and the option makes it unstable, why not deleting that option at all? If it needed some minutes I'd understand it since there is a warning in the options, but not even hours are enough.

bubnikv, 2019-08-25:
The speed of the "avoid crossing perimeters" feature has been improved with 3addae3. This issue is similar to #1756 #2040 #2458 #2546 #2631 #2783.

G-code export of your model takes 5m 40s with the "avoid crossing perimeters" enabled, while it takes 23s to export the G-code with "avoid crossing perimeters" disabled.

I am closing this issue. The "avoid crossing perimeters" is working, it could be canceled reliably. It could be faster, but it finishes in reasonable time.

I am leaving https://github.com/prusa3d/PrusaSlicer/issues/2213 open for reference of a model, which takes really long to export G-code.

closed 2019-08-25 by bubnikv