prusa3d/PrusaSlicer #1744: I can't select the extruder for different functions when wipe tower is enabled

Liger0, 2019-01-31:
### Version
1.42.0 alpha4

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
Custom printer with double extrusion and 2 separate extruders an

### Behavior
* _Describe the problem_
I can't have at the same time the purge tower and chose the extruder I want to use for different characteristics
* _Steps needed to reproduce the problem_
Enable wipe tower, and try to set an extruder on the support interface voice instead of the stock 0 option
* _Expected Results_
It should just allow it
* _Actual Results_
You can't have them both at the same time. It disables the wipe tower.
The same happens if you first selected the wanted extruder for the characteristic you want and then try to enable the wipe tower.
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

bubnikv, 2019-02-01:
If you have two separate physical extruders, then you should disable
"single extruder multi material" and the wipe tower will not be available.
Then you will not get this error message.

bubnikv, 2019-02-01:
> Also why can't it use a wipe tower? The double extruder will still need it to compensate for oozing.

Because it has not been implemented yet due to time constraints. Contributions are welcome.

> "single extruder multi material" is disabled already, and the message still shows up.

Sorry, I was wrong. It is just the wipe tower, that requires what the message says. And the wipe tower only works correctly with the single extruder multi material setup as of now. You should disable it for your dual extruder setup.

Liger0, 2019-02-03:
> 
> 
> Is it clear now? Could we close this issue?

Yeah I'm closing it, thanks.
I'll just wait and hope it will be added. :)

closed 2019-02-03 by Liger0