prusa3d/PrusaSlicer #1745: Wrong color is shown in 3D View when using "default" color

mgg4, 2019-02-01:
### Version
slic3r 1.42.0-alpha4+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 (3.51.1-1778)
Prusa MMU2 (1.0.3-212)

### Behavior
When changing extruders in the the extruder color is not properly select in some cases.
#### Scenario 1
- The "default" extruder is left unchanged on the parent in the parts tree.
- The individual parts are set to the desired extruder number directly.
- RESULT: The parts on the 3D view are shown with the correct colors assigned to each extruder.  This is the expected behavior.  If sliced, the proper extruder color is used and displayed in the Layer Preview.  

#### Scenario 2
- The "default" extruder on the parent part is set to the extruder number for the majority of the parts.
- Those parts that should be the default color have their extruder set to "default".
- Other parts are set to the desired extruder number.
- RESULT: The parts that should be shown in the desired default color are instead shown with the color assigned to extruder 1, irrespective of what extruder was actually selected for the "default".  This behavior is undesired.  If sliced, the proper extruder color is used and displayed in the Layer Preview.  

#### Expected Behavior
The color of the default extruder set in the parent part should be properly displayed in all views for child parts marked to use that default color.

This is not a new feature request.  This is a report of a malfunction of the software.

#### Project File (.3MF) where problem occurs
[Example Project.zip](https://github.com/prusa3d/Slic3r/files/2820301/Example.Project.zip)

mgg4, 2019-02-04:
Verified this problem still persists in Alpha 5.

enricoturri1966, 2019-02-05:
Working on it -> http://dev.prusa3d.com/browse/SPE-811

FidelCapo, 2019-02-06:
This issue should be fixed in next release. Closing

closed 2019-02-06 by FidelCapo