prusa3d/PrusaSlicer #1750: Failed to start slic3r 1.42.0 from git on Linux

rmoravcik, 2019-02-01:
I don't know it this is a known issue, but starting with git commit https://github.com/prusa3d/Slic3r/commit/ed9a924880a0eb12a5d0e6e29108d211da953a66 I'm not able to start self compiled slic3r on my Ubuntu 18.10.

I can see this error in terminal:
```
morgan@morgan:~/Projekty/Slic3r/Slic3r/_build/src$ ./slic3r-gui
Multisample enabled
../src/unix/glx11.cpp(581): assert ""xid"" failed in SetCurrent(): window must be shown

```

and this assert window is displayed:
[image]

This is a full backtrace output from assert window:
```
ASSERT INFO:
../src/unix/glx11.cpp(581): assert ""xid"" failed in SetCurrent(): window must be shown

BACKTRACE:
[1] wxGLContext::SetCurrent(wxGLCanvas const&) const
[2] Slic3r::GUI::GLCanvas3D::reset_legend_texture()
[3] Slic3r::GUI::Preview::load_print_as_fff()
[4] Slic3r::GUI::Preview::Preview(wxWindow*, Slic3r::DynamicPrintConfig*, Slic3r::BackgroundSlicingProcess*, Slic3r::GCodePreviewData*, std::function<void ()>)
[5] Slic3r::GUI::Plater::priv::priv(Slic3r::GUI::Plater*, Slic3r::GUI::MainFrame*)
[6] Slic3r::GUI::Plater::Plater(wxWindow*, Slic3r::GUI::MainFrame*)
[7] Slic3r::GUI::MainFrame::init_tabpanel()
[8] Slic3r::GUI::MainFrame::MainFrame()
[9] Slic3r::GUI::GUI_App::OnInit()
[10] wxEntry(int&, wchar_t**)
[11] main
[12] __libc_start_main
[13] _start

```

I'm compiling slic3r using wxWidgets 3.1 with GTK2 support.

Teteros, 2019-02-02:
Same issue on Arch here.

enricoturri1966, 2019-02-04:
This issue should be fixed by: https://github.com/prusa3d/Slic3r/commit/b3c38812826c68a15bd73ef5aba872c2d5ecead0

rmoravcik, 2019-02-04:
I can confirm, it's fixed now. Thank you