prusa3d/PrusaSlicer #1751: Scaling problem

Mack1945, 2019-02-01:
### Version
1.42.0 Alpha Slic3r Prusa Edition win64-full-201901111426_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Original Prusa I3 MK3 / , firmware 3.5.1 
* _Describe the problem_
Trying to rescale an part (object Manipulation) size = X 28 Y 140 Z 6.95 mm, the Info size = X 28 Y 140 Z 7.15 mm which size is the correct size? 

with Slic3rPE 1.41.2 when I rescale the part  the Info size = X 28 Y 140 Z 6.95 mm
*rescale any  part on the plater_
  * _If this is a command-line slicing issue, include the options used_NO
* _Expected Results_ see above statement Describe the problem
* _Actual Results_ see above statement Describe the problem
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_ NO

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-02-04:
> I also observed problems with scaling in alpha 4. When I compiled and
> tried the current master ( e59a10e
> ), the scaling worked as expected for the (multipart) model I tried it on.

I believe this issue is a duplicate of #1715, which has been fixed on the
master. Alpha5 is due today.

ruedli, 2019-02-05:
> Is it fixed for you in

It was fixed in master that I compiled (see above for the id), travelling
now with no access to pc on which I can install thing will test and let you
know later this week.

bubnikv, 2019-02-05:
Great, closing.

closed 2019-02-05 by bubnikv

ruedli, 2019-02-06:
I now installed alpha 5 and tested. As Mack1945 mentioned: it is fixed and works.