prusa3d/PrusaSlicer #1752: Super narrow Brim and Skirt paths

rtideas, 2019-02-01:
### Version
1.42.0 Alpha 4

### Operating system type + version
MacOS 10.14.2

### 3D printer brand / version + firmware version (if known)
i3 Clone - Repetier 1.0.2

### Behavior
Super narrow brims and skirts paths

Using 0.6mm nozzle and PETG, brims and skirts are very narrow. Too narrow to print without issue. 
[image]

Is there an an advance setting I may demising for brim or skirt width?

bubnikv, 2019-03-01:
Would you please attach your configuration, or the 3MF file? Thanks.

bubnikv, 2019-04-13:
No response, closing

closed 2019-04-13 by bubnikv