prusa3d/PrusaSlicer #1753: ctrl-A doesn't work well

supermerill, 2019-02-02:
### Version
1.32.0 alpha4

problem: i have to use ctrl-maj-a instead of ctrl-a in a text field to "select all"

if i comment 
```
  wxMenuItem* item_select_all = append_menu_item(editMenu, wxID_ANY, _(L("&Select all")) + "\tCtrl+A", _(L("Selects all objects")),
             [this](wxCommandEvent&) { std::cout<<"select all plater?\n"; m_plater->select_all(); }, "");
        editMenu->AppendSeparator();
        wxMenuItem* item_delete_sel = append_menu_item(editMenu, wxID_ANY, _(L("&Delete selected")) + "\tDel", _(L("Deletes the current selection")),
            [this](wxCommandEvent&) { m_plater->remove_selected(); }, "");
        wxMenuItem* item_delete_all = append_menu_item(editMenu, wxID_ANY, _(L("Delete &all")) + "\tCtrl+Del", _(L("Deletes all objects")),
            [this](wxCommandEvent&) { m_plater->reset(); }, "");

         Bind(wxEVT_UPDATE_UI, [this](wxUpdateUIEvent& evt) { evt.Enable(can_select()); }, item_select_all->GetId());
```
in MainFrame.cpp line 327, it works again.

It seems like the event is consumed, or the switch from "plater" to "setting" doesn't deactivate this menu item or something something.

reloxx13, 2019-02-02:
eg. startcode gcode, cannot ctrl+a anymore. mark all with mouse works but is annoying.

bubnikv, 2019-02-03:
The accelerators Ctrl+A, Ctrl+Del and Del were incorrectly captured 
globally by being defined in the Edit menu.
These accelerators are now suppressed in the menu (shown on Windows
but inactive, not shown on OSX / Linux),
and they are now captured by the 3D scene widget instead.

Fixed with 0c1f750cba3686ab0db19779acb8782ed047de31
Closing

In addition, following was sent to wxWidgets-users list:

We have an issue with the menu accelerators shown in the main menu. If we add a menu accelerator (for example by attaching "\tCtrl+A" to the menu text), the Ctrl+A is captured globally, so Ctrl+A cannot be used in text fields. We do capture the Ctrl+A in our 3D scene widget to select all 3D objects. 

We have found an ugly workaround, which works on Windows adding a non-breaking space 
"\t\xA0" + "Ctrl+A"
convinces wxWidgets (or maybe Windows?) to show the hot key, but to not capture it.

The hack does not work on OSX or Linux, where a warning is emitted to the console and no shortcut is shown.

I think it would be a worthy feature to allow showing the hotkey, but to not capture it. Is it feasible on other platforms than Windows?

closed 2019-02-03 by bubnikv

bubnikv, 2019-02-03:
By the way, is Cmd+A used on OSX for whole line selection?