prusa3d/PrusaSlicer #1755: Pulldown menus did not work after initial installation

Karma4Good, 2019-02-02:
### Version
1.42.0 alpha 4

### Operating system type + version
MacOS 10.14.2

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3

### Behavior
Pulldown menus at top right of screen (and also on Print Settings tab) did not initially render. Pointer position was apparently recorder however, since selecting a point in space under the menu would select an "invisible" menu item.
No idea what changed from initial attempt, but the pulldowns were visible about 10 minutes later. App was running the entire time.

Low priority but something to be aware of

bubnikv, 2019-02-03:
We have never seen such an issue. A screenshot would be very helpful.

Karma4Good, 2019-02-04:
> I was able to reproduce it on OSX Mojave with the 1.42.0-alpha5.
> It happens after the wizard finishes, it happens on a fresh install, and
> it also happens after a profile update.

Thanks, yes, it was on a fresh install when it happened. Glad you were able
to reproduce it.

Also, maybe something related: clicking on the arrow causes the menu to
drop down, but clicking on the text next to the arrow does not. I'm used to
just clicking on the name of drop-down instead of having to target the
arrow.

Thanks!

FidelCapo, 2019-02-07:
Created 817 at JIRA

bubnikv, 2019-03-01:
This will be fixed in the upcoming Slic3r 1.42.0-alpha7. Closing.

closed 2019-03-01 by bubnikv