prusa3d/PrusaSlicer #1756: Avoid crossing perimeters crashes slic3r

reloxx13, 2019-02-02:
### Version
Slic3rPE-1.42.0-alpha4+win64-full-201901281428

### Operating system type + version
Windows10x64

### Behavior
If "Avoid crossing perimeters" slic3r dies around 90%, does not response.
I tried to let is work, but after some hours it still got not finished and did not respond.

It works for small and simple STLs like a small logo.

I can hear my computer fan going down at those 90% progress bar, seems the calculation completly stopped. CPU load is going down, too.

#### Project File (.3MF) where problem occurs
[FilHolder_80mm_V1.zip](https://github.com/prusa3d/Slic3r/files/2824839/FilHolder_80mm_V1.zip)

[image]

[Slic3r_config_bundle.zip](https://github.com/prusa3d/Slic3r/files/2824841/Slic3r_config_bundle.zip)

bubnikv, 2019-02-03:
The "Avoid crossing perimeters" was never used at Prusa3D, it is mostly useful for leaky bowden extruders. Therefure we at Prusa3D did not invest time into improving this feature's quality and performance.

You will find multiple issues marked with the "Path Planning" tag concerned with this feature.

reloxx13, 2019-02-03:
guess starting to use another slicer than, thx.

Ruhel786, 2019-02-21:
So is this issue not going to get resolved then? Trying to print Lithophanes but Slic3r takes over 90-120 mins to generate GCODE.

bubnikv, 2019-02-21:
@[Ruhel786] It is resolved by disabling "Avoid crossing perimeters"

Ruhel786, 2019-02-21:
@bubnikv I need those options to print lithophanes. Hence the issue.

bubnikv, 2019-02-21:
>  @bubnikv <https://github.com/bubnikv> I need those options to print
lithophanes. Hence the issue

Why? Why do you need it on lithophanes?

Ruhel786, 2019-02-21:
@bubnikv I'm new to printing lithophanes and the guide on YT states to toggle that setting "on" to reduce filament blobs on the outside.

rennne, 2019-03-20:
With flexible filaments "Avoid crossing perimeters" is needed to get good results, even on the prusa printer.

It seems to work on smaller prints.
I had a big print yesterday and did not get the slicer to finish with "Avoid crossing perimeters" switched on

bubnikv, 2019-08-25:
Initial phase of the "avoid crossing perimeters" feature has been fixed with 3addae3, and the feature is now also significantly faster.

It takes 28s to export the G-code on my computer with the "Avoid crossing perimeters" feature enabled, while it takes 4s to export the G-code with the "Avoid crossing perimeters" feature disabled.

Closing.

closed 2019-08-25 by bubnikv