prusa3d/PrusaSlicer #1757: Not allow elephant compensation to shrink parts thinner than 2 layer lines.

oleost, 2019-02-02:
### Version
1.41.2+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa mk2s

### Behavior
When elephant compensation is used, it should not be allowed to shrink parts that are less than 2 layers wide. This causes issues printing some thin parts.

Se example pictures:

1. layer with elephant comp turned off:
[image]

1. layer with 0.33mm elephant comp:
[image]

Orignal 3d model:
[image]

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
[livspuck.zip](https://github.com/prusa3d/Slic3r/files/2824850/livspuck.zip)

supermerill, 2019-02-02:
Why you can't/don't want to lower the value? 
Are there a clear benefit to change the behavior (simple->complicated)?

bubnikv, 2019-02-03:
This is a known issue. The elefant foot compensation should be smart enough to not shrink single line or single perimeter features so much that they disappear.

This bug is a reason, why at Prusa3D we are not enabling the feature by default.

oleost, 2019-02-03:
@bubnikv Couldnt find any open Issues so I figured I create one.

probonopd, 2019-03-02:
I second this. With Elephant foot compensation enabled, this

[image]

becomes this

[image]

which results in an unprintable object because the second layer will be printed in mid air.

The solution could be to restrict elephant foot compensation to the __outermost perimeter__ of the model.

supermerill, 2019-03-02:
> The solution could be to restrict elephant foot compensation to the outermost perimeter of the model.

It is. It just reduces the "size" of the polygon by a fixed amount.

@bubnikv maybe if we do a 
union(offset(poly, -efc), diff_ex(poly, offset2_ex(poly, -ext_peri_width-efc, ext_peri_width+efc), true))
instead of 
offset(poly, -efc)
it should "resolve" this issue. But it's not very pretty, as it's a brutal threshold.

probonopd, 2019-03-02:
With "detect thin walls" on I get this:

[image]

See how the honeycomb is no longer connected to the rest?

Also not very printable.

supermerill, 2019-03-02:
> See how the honeycomb is no longer connected to the rest?

that's a thin wall problem, i'm working on it.

> Also not very printable.

why?

probonopd, 2019-03-02:
> why?

Probably because the 1-line honeycomb can easily be ripped out. Dunno.

supermerill, 2019-03-03:
I already used a one-line pattern for some "mask" i needed. With a good heated PEI surface, i had no problem (90% of success). If you need a thicker first layer for the pentagons, maybe you have to modify the object geometry.

the setting is here for the one (like me) that print with a too thin first layer, and so the width is a bit too large. The setting allow the slic3r to take that into account. If your model is too thin for th thick first width, you have to modify your model or print with a thinner first layer width (nozzle at the right height, lower first layer width).

bubnikv, 2019-03-04:
@supermerill

> union(offset(poly, -efc), diff_ex(poly, offset2_ex(poly,
-ext_peri_width-efc, ext_peri_width+efc), true))
> But it's not very pretty, as it's a brutal threshold.

I am not quite excited about this solution, but I don't know any better
solution as of now.

The problem of this solution, apart from being numerically intensive (not a
problem for a single layer) is that it may not apply the elephant foot
compensation to some protrusions, where it could. It will likely not make
the situation worse, so I will likely try it.

bubnikv, 2019-03-06:
This is not working well: 
First, the thin parts get detached from the large part.
Second, the operation
 diff_ex(poly, offset2_ex(poly, -ext_peri_width-efc, ext_peri_width+efc), true)
creates thin slits around the contour of poly even where it should not. I am afraid this is not a robust solution.

supermerill, 2019-03-07:
ah, yes. my bad.

Maybe a special rule for thin wall in the first layer if this parameter is activated? i.e, reduce the threshold for minimum width to (firstlayerwidth - elephant * 2).  I will do some testing on that.

bubnikv, 2020-01-02:
The elephant foot compensation improvements are part of PrusaSlicer 2.2.0-alpha2. Closing.

closed 2020-01-02 by bubnikv

probonopd, 2020-01-04:
This rocks.

bubnikv, 2020-01-04:
Yeah, there are features, that one writes for himself :-)
I hate to scrap the elephant foot off, and the current feature allows one to set it and forget it.

JohnOCFII, 2020-01-04:
It is working well in my testing too!

bitingsock, 2022-08-16:
I would make the point that with the inclusion of arachne this behavior should be opened pack up. Consolidating to a single line is useful sometimes and arachne can make it wider to split the difference.
Personally, I don't like the limitation and prefer to look for these problems on my own, though I suppose most others would not.
Moreover, this limitation was never forced on normal XY compensation and thus seems arbitrary.