prusa3d/PrusaSlicer #1758: Atribution of stl.icns and gcode.icns ommited from README.md

reven, 2019-02-03:
### Version
Not version relevant

### Operating system type + version
MacOS (not version relevant)

### 3D printer brand / version + firmware version (if known)
not relevant

### Behavior
* The really nice STL and GCODE icons that are used in MacOS were originally designed by Akira Yasuda. I originally added them to Slic3r (https://github.com/slic3r/Slic3r/pull/4376) and they were pulled into Slic3rPE (https://github.com/prusa3d/Slic3r/commit/c0df7f343ee28ccd96185a6599da8d848f692d35), but somehow the acknowledgement to Akira that I added to the README.md was not.

I'd like to see Akira's work recognized, please.

bubnikv, 2019-02-03:
That was surely not our intention.

We should add the attributions to the GUI somewhere, including all the 3rd party open source libraries we are using.

bubnikv, 2019-05-17:
Atrubtion
[image]

Closing

closed 2019-05-17 by bubnikv