prusa3d/PrusaSlicer #1760: Preset bundle PrusaResearch version not found crash

drdelaney, 2019-02-03:
### Version
Slic3rPE-1.42.0-alpha3+win64-full-201901111426
Slic3rPE-1.42.0-alpha4+win64-full-201901281428
Both from the released zip files.

### Operating system type + version
Windows 10 1803 build 17134.523

### 3D printer brand / version + firmware version (if known)
i3 Mk2 Clone (But this does't matter)

### Behavior
In both 1.42.0alpha3 and alpha4, I get a popup (sometimes it is a pop up diaglog that doesn't show any data) that shows:
Preset bundle `PrusaResearch` version not found in index: 0.3.4

Upon hitting ok, the app seems to crash or at least close.

drdelaney, 2019-02-03:
Note this was detected on an upgraded 1.41.2 config directory.  If I run the binary without an existing configuration path (--datadir) it seems to open ok.

This seems to also go away when I delete the "vendor" folder.  It re-creates it and is happy.

drdelaney, 2019-02-03:
Even after the vendor folder is removed, and the app opens up, restarting the app causes it to display an error and crash.  Noticed this when looking at #1761 , perhaps this error box is the ini error and closing afterwards.  Cannot verify this at this time.

bubnikv, 2019-02-03:
I suppose it is the same as #1760, which is hopefully fixed now on the master. Closing.

closed 2019-02-03 by bubnikv

grafzahl, 2019-02-12:
Hey there, i currently get this error when i tried to start the current Alpha5:
`Preset bundle `PrusaResearch` version not found in index: 0.4.0-alpha3`

Right now my old Alpha4 also drops the same error and i dont even know whats causing this.

vojtechkral, 2019-02-12:
@grafzahl There was a mixup between the stable and alpha versions. To fix it, please remove either the whole configuration directory or the `cache` subdirectory inside the config directory.

OmiPops, 2019-02-13:
Thanks - removing the cache directory worked for me - MAC OS 10.14.3 with Slic3r Alpha5

vojtechkral, 2019-02-13:
Thanks for reporting. I'll look into making the update check process more resilient against a problem like this.

serg1669, 2026-01-28:
Hello, I also have an error when I launch the program. Please advise me on how to remove the error. Let me tell you that when I click the OK button, the program continues to work normally.

[image]