prusa3d/PrusaSlicer #1761: desktop.ini still displaying error

drdelaney, 2019-02-03:
### Version
Slic3rPE-1.42.0-alpha3+win64-full-201901111426
Slic3rPE-1.42.0-alpha4+win64-full-201901281428
Both from the released zip files.

### Operating system type + version
Windows 10 1803 build 17134.523

### 3D printer brand / version + firmware version (if known)
i3 Mk2 Clone (But this does't matter)

### Behavior
Even after the closing of #1298 , using both alpha3 and 4, I am getting a error popup:

Vendor Config Bundle `desktop` is not valid: Missing secion or key: `vendor`.

This goes away after I remove the desktop.ini files.
I have verified these files are marked as hidden AND read only.

drdelaney, 2019-02-03:
I realized this just went away when I was looking into #1760 , deleting the vendor folder made this error go away.  Perhaps the check is not being ran on all imported configurations folders?

bubnikv, 2019-02-03:
You are right, I fixed it just for the preset directories in #1760. 
Now I hope I covered all file enumeration loops with f9743d17e9b700e5d6413de2d0eb497fc283c4f4

Closing.

closed 2019-02-03 by bubnikv