prusa3d/PrusaSlicer #1762: save variable layer heights?

LaneTee, 2019-02-03:
### Version
Slic3r Prusa Edition 1.41.2(64 bit)
### Operating system type + version
Windows 10
### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk2/3
### Behavior
Saving a configuration does NOT save custom layer settings. 
This is a new feature request--save these settings along with the other config information so I don't have to re-enter the layer height settings every time.

bubnikv, 2019-02-03:
It should be getting saved into a 3MF file. Does it not work for you?

LaneTee, 2019-02-05:
How do I export/import a 3MF file in Prusa/Slic3r? I assumed the .ini file
would contain the layer data.

Thank you for your help.

Lane Hauck

FidelCapo, 2019-02-06:
At top menu: File - Save Project or Ctrl-S.

bubnikv, 2019-02-06:
On Slic3r 1.41.2 and older, it is menu Plater -> Export as 3MF

LaneTee, 2019-02-06:
I set custom layer heights with settings/layers and exported to a 3MF file,
then imported the 3MF file and the settings/layers table was empty.

--Lane

bubnikv, 2019-05-04:
so it is about the tabular entry, not about the smooth variable layer height?

LaneTee, 2019-05-04:
Yes, the table I see under Settings\Layers

bubnikv, 2019-05-04:
thanks. I have a bad news for you, as for now the upcoming 1.42.0 release does not support the layer height by table feature at all. We plan to implement it in a richer way in the future, by supporting all parameter overrides for layer spans.

LaneTee, 2019-05-04:
Pity. I find the table more precise and easier to use than dragging around
a curve. I'll keep my current release until the future version supports it
again.

lukasmatena, 2019-08-26:
> until the future version supports it again

Your wish has been fulfilled with PrusaSlicer2.1.0. Closing the issue.

closed 2019-08-26 by lukasmatena