prusa3d/PrusaSlicer #1766: Object copy moved on layer 18 in multiobject print corrupting print

anton-matosov, 2019-02-04:
### Version
1.41.2+

### Operating system type + version
macOS 10.14.2 

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3, stock with FW 3.5.1-1778

### Behavior
* One of the object's copy moved on layer 18 in multiobject print corrupting the print
Steps (approximate, can't reproduce):
1. Add STL object to print
1. Scale to 50%, slice and export
1. Add 1 more copy
1. Arrange and slice
1. Scale to 65%
1. Arrange and export
1. Add 2 more copies
1. Slice and export

#### Project File (.3MF) where problem occurs
gcode - [FAILED-Elektriciteitsmast-.70-x4_0.2mm_PLA_MK3.gcode.zip](https://github.com/prusa3d/Slic3r/files/2826595/FAILED-Elektriciteitsmast-.70-x4_0.2mm_PLA_MK3.gcode.zip)
source STL - [Electricity_pole.zip](https://github.com/prusa3d/Slic3r/files/2826546/Electricity_pole.zip)
I do not have .3MF file with issue as I have discovered issue after print have finished
[image]

FidelCapo, 2019-02-04:
Thank you for report, but I can not reproduce this error with either version 1.41.2 or version 1.42.0-alpha4. Can you please attach 3MF if you will see this issue again? Thanks

bubnikv, 2019-02-04:
I can confirm the G-code is really broken. I wonder how it happened.

Did you see the shifted G-code in the final G-code preview in Slic3r? It should be visible there as well.

anton-matosov, 2019-02-06:
Unfortunately I have not checked the gcode preview after I've added more copies of the model and just exported gcodes and sent it to print, and didn't even save 3mf... 
Lesson learned, I will always save 3mf for all my prints and will definitely review gcode before printing.

bubnikv, 2019-03-12:
Isn't it the same issue as described here?
https://www.facebook.com/groups/prusacommunity/permalink/958562104484522/

Were you moving the objects during G-code export?

anton-matosov, 2019-03-15:
I can’t see the post, group is not public, but I’ll not exclude this possibility. I might have clicked arrange after initiating slicing/export

bubnikv, 2019-04-13:
I am closing this issue, as the 1.42.0 release should fix this issue.

closed 2019-04-13 by bubnikv