prusa3d/PrusaSlicer #1767: [1.42 New UI] Tab key in the object settings panel selects delete button for setting but it is not highlighted

lf-, 2019-02-04:
### Version
_Version of Slic3r Prusa Edition used goes here_
1.42.0-alpha4+win64

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 1809
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### Behavior
* _Describe the problem_
When tabbing through object settings, the X button is selected but there is no visible highlight.

* _Steps needed to reproduce the problem_
Insert an object into slic3r, right click it, select Infill or another setting, then click in one of the Object Settings boxes. Press tab and observe that the text cursor has disappeared and it is unclear what is selected.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
[image]
or
[image]

* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

[image]

_Is this a new feature request?_
Bug.

lf-, 2019-02-04:
Another bug is that the Delete key is globally captured even if you are in one of those boxes, deleting your model irreversibly rather than editing the text. Expected user reaction: (┛ಠ_ಠ)┛彡┻━┻

bubnikv, 2019-02-04:
The Delete / Control+Delete global capture is already fixed on the master.

YuSanka, 2019-02-14:
Thanks for your report.
Working on it in [JIRA-826](http://dev.prusa3d.com/browse/SPE-826)

bubnikv, 2019-08-08:
Highlight of the delete button will be implemented by the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-08 by bubnikv