prusa3d/PrusaSlicer #1768: Auto-change of infill causes crash

swissfreek, 2019-02-04:
### Version
Slic3rPE-1.42.0-alpha4+full-201901281426

### Operating system type + version
MacOS Mojave 10.14.3
In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2.5 firmware 3.5.1-1778

### Behavior
* I usually use Gyroid infill, around 20%. If I set infill to 100% for a part from the Plater view, Slic3r says "Gyroid infill not supported for that infill percentage, would you like to switch to Rectilinear?". When I click either Yes or No, Slic3r immediately crashes to desktop. This crash does NOT happen when you are in the Print Settings tab. In the Print Settings tab, the dialog pops up, you click Yes, and it changes and everything works fine, or you click No and it doesn't change the infill percentage. But in the Plater tab, clicking Yes or No results in a crash. It does not matter whether there is an object loaded or it's just a blank build plate.
* Set Gyroid infill with a percentage less than 100%. From the Plater view, set infill to 100%. Click Yes or No in the dialog when it asks if you want to switch to Rectilinear. Instant crash.

[image]
[image]
[Slic3r Crash Log.txt](https://github.com/prusa3d/Slic3r/files/2829072/Slic3r.Crash.Log.txt)

FidelCapo, 2019-02-06:
Thank you for report. I can reproduce this issue and crash.
Created 815 in JIRA.

FidelCapo, 2019-02-20:
This issue should be fixed in next release. Closing.

closed 2019-02-20 by FidelCapo