prusa3d/PrusaSlicer #1770: assert ""pos <= m_rangeMax"" failed in SetValue(): invalid value in wxGauge::SetValue()

rmoravcik, 2019-02-04:
Hi all,

I was increasing a number of instances (using + button from the top menu) of this simple object and this assert was raised:

[image]

```
ASSERT INFO:
../src/gtk/gauge.cpp(94): assert ""pos <= m_rangeMax"" failed in SetValue(): invalid value in wxGauge::SetValue()

BACKTRACE:
[1] wxGauge::SetValue(int)
[2] Slic3r::GUI::Plater::priv::on_slicing_update(Slic3r::SlicingStatusEvent&)
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxEvtHandler::ProcessPendingEvents()
[9] wxAppConsoleBase::ProcessPendingEvents()
[10] wxEventLoopBase::DoYieldFor(long)
[11] wxGUIEventLoop::DoYieldFor(long)
[12] wxEventLoopBase::YieldFor(long)
[13] void libnest2d::selections::_FirstFitSelection<ClipperLib::PolygonImpl>::packItems<libnest2d::placers::_NofitPolyPlacer<ClipperLib::PolygonImpl, libnest2d::_Box<ClipperLib::IntPoint> >, __gnu_cxx::__normal_iterator<std::reference_wrapper<libnest2d::_Item<ClipperLib::PolygonImpl> >*, std::vector<std::reference_wrapper<libnest2d::_Item<ClipperLib::PolygonImpl> >, std::allocator<std::reference_wrapper<libnest2d::_Item<ClipperLib::PolygonImpl> > > > >, libnest2d::_Box<ClipperLib::IntPoint>&, libnest2d::placers::NfpPConfig<ClipperLib::PolygonImpl>&>(__gnu_cxx::__normal_iterator<std::reference_wrapper<libnest2d::_Item<ClipperLib::PolygonImpl> >*, std::vector<std::reference_wrapper<libnest2d::_Item<ClipperLib::PolygonImpl> >, std::allocator<std::reference_wrapper<libnest2d::_Item<ClipperLib::PolygonImpl> > > > >, __gnu_cxx::__normal_iterator<std::reference_wrapper<libnest2d::_Item<ClipperLib::PolygonImpl> >*, std::vector<std::reference_wrapper<libnest2d::_Item<ClipperLib::PolygonImpl> >, std::allocator<std::reference_wrapper<libnest2d::_Item<ClipperLib::PolygonImpl> > > > >, libnest2d::_Box<ClipperLib::IntPoint>&, libnest2d::placers::NfpPConfig<ClipperLib::PolygonImpl>&)
[14] Slic3r::arr::arrange(Slic3r::Model&, int, Slic3r::Polyline const&, Slic3r::arr::BedShapeHint, bool, std::function<void (unsigned int)>, std::function<bool ()>)
[15] Slic3r::GUI::Plater::priv::arrange()
[16] Slic3r::GUI::Plater::increase_instances(unsigned long)
[17] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[18] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[19] wxEvtHandler::TryHereOnly(wxEvent&)
[20] wxEvtHandler::ProcessEventLocally(wxEvent&)
[21] wxEvtHandler::ProcessEvent(wxEvent&)
[22] wxEvtHandler::ProcessPendingEvents()
[23] wxAppConsoleBase::ProcessPendingEvents()
[24] wxApp::DoIdle()
[25] g_main_context_dispatch
[26] g_main_loop_run
[27] gtk_main
[28] wxGUIEventLoop::DoRun()
[29] wxEventLoopBase::Run()
[30] wxAppConsoleBase::MainLoop()
[31] wxEntry(int&, wchar_t**)
[32] main
[33] __libc_start_main
[34] _start
```

I'm using self-compiled git version (87ae0610619e296fb7c4ed74ed7b7a28284bce33). I'm using GTK2 version of wxWidget 3.1 running on Ubuntu Linux 18.10.

[Part Studio 1 - Part 1.zip](https://github.com/prusa3d/Slic3r/files/2829459/Part.Studio.1.-.Part.1.zip)

bubnikv, 2019-04-13:
do you still have the issue with the current master?

rmoravcik, 2019-04-13:
Yes, I'm still able to reproduce it. I'm using (255a4e05dca6b227c7880ae369823e2c92990d1d).

[image]

```
ASSERT INFO:
../src/gtk/gauge.cpp(94): assert ""pos <= m_rangeMax"" failed in SetValue(): invalid value in wxGauge::SetValue()

BACKTRACE:
[1] wxGauge::SetValue(int)
[2] Slic3r::GUI::Plater::priv::on_slicing_update(Slic3r::SlicingStatusEvent&)
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxEvtHandler::ProcessPendingEvents()
[9] wxAppConsoleBase::ProcessPendingEvents()
[10] wxEventLoopBase::DoYieldFor(long)
[11] wxGUIEventLoop::DoYieldFor(long)
[12] wxEventLoopBase::YieldFor(long)
[13] void libnest2d::selections::_FirstFitSelection<ClipperLib::Polygon>::packItems<libnest2d::placers::_NofitPolyPlacer<ClipperLib::Polygon, libnest2d::_Box<ClipperLib::IntPoint> >, __gnu_cxx::__normal_iterator<std::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> >*, std::vector<std::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> >, std::allocator<std::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> > > > >, libnest2d::_Box<ClipperLib::IntPoint>&, libnest2d::placers::NfpPConfig<ClipperLib::Polygon>&>(__gnu_cxx::__normal_iterator<std::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> >*, std::vector<std::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> >, std::allocator<std::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> > > > >, __gnu_cxx::__normal_iterator<std::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> >*, std::vector<std::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> >, std::allocator<std::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> > > > >, libnest2d::_Box<ClipperLib::IntPoint>&, libnest2d::placers::NfpPConfig<ClipperLib::Polygon>&)
[14] Slic3r::arr::arrange(Slic3r::Model&, int, Slic3r::Polyline const&, Slic3r::arr::BedShapeHint, bool, std::function<void (unsigned int)>, std::function<bool ()>)
[15] Slic3r::GUI::Plater::priv::arrange()
[16] Slic3r::GUI::Plater::increase_instances(unsigned long)
[17] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[18] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[19] wxEvtHandler::TryHereOnly(wxEvent&)
[20] wxEvtHandler::ProcessEventLocally(wxEvent&)
[21] wxEvtHandler::ProcessEvent(wxEvent&)
[22] wxEvtHandler::ProcessPendingEvents()
[23] wxAppConsoleBase::ProcessPendingEvents()
[24] wxApp::DoIdle()
[25] g_main_context_dispatch
[26] g_main_loop_run
[27] gtk_main
[28] wxGUIEventLoop::DoRun()
[29] wxEventLoopBase::Run()
[30] wxAppConsoleBase::MainLoop()
[31] wxEntry(int&, wchar_t**)
[32] Slic3r::CLI::run(int, char**)
[33] main
[34] __libc_start_main
[35] _start
```

bubnikv, 2019-04-13:
@YuSanka @vojtechkral would you guys look into it? Thanks

closed 2019-04-17 by vojtechkral

vojtechkral, 2019-04-17:
@rmoravcik Hopefully this is fixed now, you can test with this build:

[PrusaSlicer-1.42.0-beta2+27-vk-bugfixes-full-g6f0e208-201904171306.zip](https://github.com/prusa3d/Slic3r/files/3090192/PrusaSlicer-1.42.0-beta2%2B27-vk-bugfixes-full-g6f0e208-201904171306.zip)

commit b57452d0 (2019-04-17): Fix a bunch of race conditions between arrange and bg processing, fix #1770