prusa3d/PrusaSlicer #1771: 1.42.0-alpha5 dropbox menus randomly partially render off edge of window on linux

jhoblitt, 2019-02-04:
### Version
`Slic3rPE-1.42.0-alpha5+linux64-full-201902041815.AppImage`

### Operating system type + version

```
$ cat /etc/redhat-release 
Fedora release 29 (Twenty Nine)
```

### Behavior
* When a print setting, filament, or printer config with a long name is selected under Xorg or Xwayland the right hand side dropdown box menus sometimes, but not always, appear to be rendered partially off of the right side of the window.  Strangely, giving a dropdown box focus with the mouse then cycling through the options using the arrow keys temparilly fixes the rendering problem while background slicing runs, then it the UI returns to the glitched state.

This issue seems to be present in all prevous 1.42.x alpha releases through alpha5.

- At startup
  [image]
- Long setting names
  [image]
-  While cycling with arrow keys
  [image]

#### Project File (.3MF) where problem occurs

This is *an* example 3mf but it isn't very difficult to cause this rendering glitch to happen.

[Glasses_Case_Medium_-_Outer_Shell-top-pla.zip](https://github.com/prusa3d/Slic3r/files/2829608/Glasses_Case_Medium_-_Outer_Shell-top-pla.zip)

rhinodavid, 2019-02-05:
+1 on MacOS 10.14.3

reinhapa, 2019-02-06:
Same here on Fedora Linux every time when adding a new item and it get's restored as soon I remove all parts from the plate.

On my  Mac I could not reproduce the behaviour.

YuSanka, 2019-02-07:
Working on it in http://dev.prusa3d.com/browse/SPE-821

bubnikv, 2019-03-01:
In upcoming Slic3r 1.42.0-alpha7, the width of the side panel should be derived from the font size. That should fix most of the issues.

There is an issue with how GTK handles texts longer than the combo box and how the text overflows to the left, but that is yet another issue.

Closing.

closed 2019-03-01 by bubnikv

reinhapa, 2019-03-01:
Looking forward to test it then 👍