prusa3d/PrusaSlicer #1772: [1.42.0-alpha5+win64] 3D editor view broken

yagona, 2019-02-04:
### Version
Slic3r Prusa Edition
Version:   1.42.0-alpha5+win64
Build:     Slic3rPE-1.42.0-alpha5+win64-full-201902041815

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17763), 64-bit edition

OpenGL installation
GL version:   4.6.0 NVIDIA 417.71
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1060 6GB/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
[... 323 lines cut ...]
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
modified Anycubic i3 Mega
TMC2208, Titan Extruder, E3D V6 bowden, 0.35mm nozzle
modified Marlin bugfix-1.1.x from 24.01.2019

### Behavior
* _Describe the problem_
3D editor view not working (see screenshot). I suppose that the colorful regions result from the arrows of the coordinate system. Zooming out is not possible.

* _Steps needed to reproduce the problem_
The problem occurs directly after starting the software.

* _Expected Results_
Working 3D editor view.

* _Actual Results_
[image]

_Is this a new feature request?_
No.

enricoturri1966, 2019-02-05:
Working on it: http://dev.prusa3d.com/browse/SPE-809

FidelCapo, 2019-02-07:
This issue should be fixed in next release. Closing

closed 2019-02-07 by FidelCapo