prusa3d/PrusaSlicer #1773: [alpha5] Importing config bundle creates new "user defined printers"

Boby71, 2019-02-04:
### Version
1.4.2-alpha5

### Operating system type + version
WIN10/64 Bit

### 3D printer brand / version + firmware version (if known)
i3MK2.5

### Behavior
I did the following:
1) Exported a config bundle from the alpha4
2) I installed the alpha5
3) I started a "Configuration wizzard" for my MK2.5 (scrath option)
4) I imported the alpha4 config bundle into alpha5

After this, I see some new "user defined" printers I never had before.

[image]
The "Original Prusa i3 MK2.5 - Custom" was the one and only user defined printer for me before

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[2019-02-04 23_46_18-.zip](https://github.com/prusa3d/Slic3r/files/2830094/2019-02-04.23_46_18-.zip)
No project file, as the problem occurs before any object is being loaded. But I added the config bundles of the alpha4 and alpha5

supermerill, 2019-02-06:
What i understand: 

* system presets are from the ones from the "vendor" folder, filled by the configuration wizard.

* You used the "import config bundle", so slic3rPE think it's a bundle of user-defined config.

bubnikv, 2019-02-07:
The explanation is, that the MK2 MMU1 profiles were renamed at Prusa3D
system profiles, therefore if you import an old config bundle into a new
slic3r, it considers these configs with old names to be non-system, as
their names do not match the new system names. If the imported profiles had
matched the system names, these imported profiles would just be ignored.

This will not be a problem in the future, as the Slic3r 1.42 does not
export system profiles into the user config bundle anymore, but we need to
think at Prusa3D how to handle renaming of profiles into the future.
Namely, if you save a profile, which was derived from a system profile, you
will currently lose the link to the system profile once we rename the
system profile from which you derived your user profile. I suppose we
should introduce a mapping from the old profile name to a new profile name
in our system profiles and keep this mapping indefinitely.

bubnikv, 2019-03-01:
Do you find my explanation reasonable? If so, would you please close this issue?
Thanks.

closed 2019-03-02 by Boby71