prusa3d/PrusaSlicer #1774: Truncated name column in list of objects

marchyman, 2019-02-04:
### Version
_Version of Slic3r Prusa Edition used goes here_

1.42.0-Alpha5

### Operating system type + version

macOS 10.14.3

### Behavior

The `name` column in the list of objects is truncated at startup.

[image]

That screen shot is from alpha4, alpha5 does the same thing.  Dragging the column separator works, but the new location is not saved meaning I've not figured out a way to save it.

bubnikv, 2019-03-01:
@vojtechkral @YuSanka would you guys look into this issue? Thanks.

bubnikv, 2019-04-13:
Duplicate of https://github.com/prusa3d/Slic3r/issues/1927

timur-tabi, 2019-05-27:
This problem still exists in the 2.0 release.

YuSanka, 2019-05-27:
Thanks for your report. But we can't reproduce this bug. Could you please  provide the exact steps you did to get such a result.

timur-tabi, 2019-05-28:
There are no steps.  It always looks like this when I start the program.  I'm running Mac OS X 10.9, and the report is from someone running 10.14, so it's not version-specific.

Here's a screen shot of what the app looks like when I first start it.  Look at the white box on the left.  The first column says only "N.."

[image]

FidelCapo, 2019-05-30:
Can you please tell us system spec like Model of your mac, if you have some external monitor and resolution of your monitor(s)? Thank you

timur-tabi, 2019-05-30:
NVIDIA GeForce 9400:

  Chipset Model:	NVIDIA GeForce 9400
  Type:	GPU
  Bus:	PCI
  VRAM (Total):	256 MB
  Vendor:	NVIDIA (0x10de)
  Device ID:	0x0861
  Revision ID:	0x00b1
  ROM Revision:	3362
  Displays:
2778X:
  Resolution:	2560 x 1440 @ 60 Hz
  Pixel Depth:	32-Bit Color (ARGB8888)
  Display Serial Number:	GJGH7HA010636
  Main Display:	Yes
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported
  Connection Type:	DisplayPort
  Television:	Yes

prowlmedia, 2019-07-17:
Same here.  Column is truncated on opening.   Needs a minimum character width of 20 or something. 
MacOS 10.13.2 
4K monitor external

marchyman, 2019-07-27:
Mohave 10.14.6 on an iMac Retina 5K, 27-inch, 2019 machine.   I see the exact same thing on a 2012 iMac 27" machine, too.  PrusaSlicer-2.0.

timur-tabi, 2019-09-27:
This problem still exists in the 2.1 release.

YuSanka, 2019-09-27:
> This problem still exists in the 2.1 release.

Yes, its bug fix will be a part of next release. As a mentioned in #2909, [here](https://drive.google.com/open?id=1YQq_E24Ixsmk3oosGcakvMocNTJkXZK7) is a test build.

 Could you test this build too?

timur-tabi, 2019-09-27:
Yes, your test build fixes the problem for me, thanks.

ryanneufeld, 2019-11-16:
I'll add another confirmation of the fix. The linked dev build did not display the problem

timur-tabi, 2019-12-14:
This problem still exists in 2.1.1+, which I just downloaded.  Why didn't the fix make this release?

ryanneufeld, 2019-12-14:
@YuSanka This did not make it in the latest release it seems.

lukasmatena, 2019-12-14:
2.1.1. is a minor update of 2.1.0. Most of the new features and fixes (including this one) will be in 2.2.0.

bubnikv, 2019-12-17:
This issue will finally be fixed in PrusaSlicer 2.2. Closing.

closed 2019-12-17 by bubnikv