prusa3d/PrusaSlicer #1776: [Request] Rename Object for new Octoprint-Cancelobject feature

andrelec1, 2019-02-05:
### Version
Slic3rPE-1.42.0-alpha5+win64-full-201902041815

### Operating system type + version
Windows 10 

I love the new support for Octoprint-Cancelobject ...

But in most case, i print several instance of the same object, and its hard to identify who is where.

My first try named my objects like that  :
```
rail bottom.stl id:0 copy 0
rail bottom.stl id:0 copy 1
rail bottom.stl id:0 copy 2
```
then i try to separate the object but that same ...  :
```
rail top.stl id:1 copy 0
rail top.stl id:2 copy 0
rail top.stl id:3 copy 0
```

It would be nice if we can add custom label or rename objects.

paukstelis, 2019-02-05:
This would be pretty useful. I just glanced through and it looks it wouldn't be all that hard. Need to provide a set_name function for ModelObject and come up with the front end interface for accessing that? I might take a whack at it.

bubnikv, 2019-02-05:
It is already there. Just click on the name at the side panel and start
typing.

andrelec1, 2019-02-05:
Hoo, i didn't know that feature exist ...
btw this not work on instance of same objet ...
( but, i can separate object the naming, so that all i need ^^ )

This function just need a better UX ^^

paukstelis, 2019-02-05:
Woah. That's awesome! Thanks for pointing that out.

supermerill, 2019-02-06:
> This function just need a better UX ^^

The octoprint plug-in may be able in the future (if someone has the motivation, time & skill to  do the work) to show the first layer and to select the part to remove visually by clicking on it. With that, no more need for "names". See https://github.com/paukstelis/Octoprint-Cancelobject/issues/19

paukstelis, 2019-02-08:
I'm making some progress on integrating the plugin with the GcodeViewer to allow cancelling directly. The problem I am running into (that isn't an issue when just using a list) is that any adhesion objects (skirt, brim, etc.) get included in one of the objects I track extrusion moves. One thought would be to tag these as a separate object, another option might be to parse all extrusion moves on file upload and store the objects' center-of-mass in XY in some comment in the file.

supermerill, 2019-02-09:
it's easy to crate a comment block with center of mass and a aabb boundingbox. I don't know for the other slicers.

paukstelis, 2019-02-09:
Sure. Parsing those after file upload will avoid having to track extrusion like I'll need to do with other slicers.  A comment with CoM and object name would be ideal.

supermerill, 2019-02-09:
something like:
```
; boundingbox "rail top.stl" id:1 copy 0 center:[23.7358,12.1248,12.49248] boundingbox:[100,100,10]
; boundingbox "rail top.stl" id:1 copy 1 center:[123.7358,12.1248,12.49248] boundingbox:[100,100,10]
```
with center the center of the bounding box?

or maybe i should add the real center of mass of the object?

```
; boundingbox "rail top.stl id:1 copy 0" object_center:[23.7358,12.1248,12.49248] boundingbox_center:[25,12,5] boundingbox_size:[100,100,10]
```

or maybe i should go full json?
```
; boundingbox:{"name":"rail top.stl id:1 copy 0", "object_center":[23.7358,12.1248,12.49248], "boundingbox_center":[25,12,5], "boundingbox_size":[100,100,10] }
```

paukstelis, 2019-02-09:
json makes things a bit easier. I think the object center is more useful than the bounding box center, but its not going to hurt to have both.

supermerill, 2019-02-09:
without the bb center, you can't place the bb.

paukstelis, 2019-02-12:
> without the bb center, you can't place the bb.

I was planning on placing it at the object center, though it really doesn't matter which is used.

I also have the first integration with the gcode viewer that marks each object with a circle that can be clicked to allow cancelling. Use the refresh button to update positions of objects in the gcode viewer. Use reset button if things like brims make the objects positions way off on the first couple of layers.

Its in the positiontrack branch or install from zip:
https://github.com/paukstelis/Octoprint-Cancelobject/archive/positiontrack.zip

supermerill, 2019-02-12:
nice timing, you can grab the experimental function on my fork here: https://github.com/supermerill/Slic3r/releases/tag/1.42.0-alpha5.0

bubnikv, 2019-04-13:
Is there anything to be done at Slic3r's side, or could we close this issue?

paukstelis, 2019-04-13:
Everything in the original issue has been completed and is in 1.42.

bubnikv, 2019-04-13:
@paukstelis cool, thanks.

closed 2019-04-13 by bubnikv