prusa3d/PrusaSlicer #1779: Consider rolling back to wx 3.0

SukkoPera, 2019-02-05:
### Version
1.42-alpha3 and following

### Behavior
This is a request to rollback to wx 3.0, as 3.1 isn't widespread yet and most distributions still ship with 3.0.

Actually, quoting the officiale wx website, _3.1.0 is officially a “development” version because it is not fully compatible with 3.0.x_.

Since most applications using wx still use 3.0, and as I'm not sure they can be installed side by side (any info on this), installing 3.1 just for Slic3r breaks all other applications.

At least, please postpone the version bump to a later version, as doing it between alpha2 and alpha3 is pretty subtle.

I guess this all boils down to: what features of 3.1 that are not available in 3.0 is Slic3r using?

vojtechkral, 2019-02-05:
You can build with `-DSLIC3R_WX_STABLE=1` to use wxWidgets 3.0 on Linux.

I should add this to the Linux build doc.

bubnikv, 2019-02-05:
We are currently using wxWidgets 3.1.1 on all platforms to produce our
binary builds. wxWidgets are often finicky, versions differ in subtle
details, which may be important.

wxWidgets 3.1.1 support HiDPI displays, modern OpenGL etc.

Just try to compile against wxWidgets 3.0 and test it, what kind of issues
you will find, which work correctly in our binary builds.We may then
discuss how to patch these discrepancies in the source code, so we would
support our software as a distro package dynamically linked to a distro
wxWidgets 3.0 package.

SukkoPera, 2019-02-05:
That flag definitely needs documenting!!!

The build goes fine (with -alpha5 at least) and the program starts, but then I get a pretty weird view:

[image]

Any hints?

vojtechkral, 2019-02-05:
@SukkoPera That's #1772

SukkoPera, 2019-02-06:
Thanks, patch applied. Application seems to work fine at first glance.

I will report if I find any misbehaviour. I think we can close this issue for the moment, thanks everyone.

closed 2019-02-06 by SukkoPera

SukkoPera, 2019-02-06:
By the way, as a last note, I still think it's too early for a mainly-3.1 application. It should really be the other way round: stick to 3.0 by default and have a compile switch for enabling experimental 3.1 support. Then, when everybody is using 3.1, you can make it the default.

Just my $.02.

bubnikv, 2019-02-06:
> By the way, as a last note, I still think it's too early for a mainly-3.1
application. It should really be the other way round: stick to 3.0 by
default and have a compile switch for enabling experimental 3.1 support.
Then, when everybody is using 3.1, you can make it the default.

This is a view skewed by a Linux user used to a distro distribution model.
For an application with quick update cycles, the distribution model of
Windows or OSX applications (everything distributed with the application)
is better IMHO. This is not only my view of point, even Linus Torvalds
likes the AppImage distribution model.

SukkoPera, 2019-02-06:
I don't like that approach at all, but I guess you are right, I'm linked to the past, where hard drive and memory space were at a premium and you didn't want to have 20 copies of the same library around. I guess the whole concept of shared libraries is now obsolete, isn't it?

SukkoPera, 2019-02-06:
I really don't get the point of using 3.1 when none of its features is really crucial for the application. You say it supports HiDPI, but then I see no difference whatsoever on my HiDPI display, and the app is hard to use anyway (See #1709).

vojtechkral, 2019-02-06:
@SukkoPera 

> I guess the whole concept of shared libraries is now obsolete, isn't it?

I wouldn't go that far :) Shared libs is a concept that especially under Linux works pretty well for established software with not-too-frequent releases. Such an example would be GTK, which we still link against dynamically. For faster moving and not so established targets such as Slic3r, with quite a lot of dependencies, it doesn't work that well. (It can, but then someone has to work hard to get it right in multuple distros.)

> I really don't get the point of using 3.1 when none of its features is really crucial for the application.

I believe the initial push for wx 3.1 was mainly the translation context (in gettext). At that time some strings were impossible to get translated right because of the lack of support for context in wx 3.0. However since then I believe the strings have changed and I _think_ we don't need context _at the moment_.