prusa3d/PrusaSlicer #1780: Rotation arrows in wrong directions

heczis, 2019-02-05:
### Version
1.42.0-alpha5.r1.g5207d31df

### Operating system type + version
ArchLinux

### Behavior
When typing into the text-fields in Instance manipulation, arrows appear in the 3D view, presumably to suggest the rotation direction.
These arrows seem to be swapped in the case of rotation around Y a Z axes (X is correct).
[image]
[image]
[image]

andrelec1, 2019-02-05:
[image]

No, the arrow is placed on the original orientation of the objet ... 
But i don't know if is a bug or feature ^^

heczis, 2019-02-05:
Yes, I figured it some time afterwards, too.
First thing that confuses me (as was shown by me creating this issue :-] ) is that the axes (X, Y, Z) change when rotating the object. Scaling and resizing of the object(s) (or instances) are performed along the changed (local) directions, but translation is performed in the global coordinate system. I'm OK with these.
Next is that a sequence of rotations is not commutative in general. Then what do the numbers in those three text fields mean? When an object with two instances is rotated, then selecting the parent object shows "Rotation: 0 0 0", but selecting any of the instances shows some fuzzy numbers.
There's another effect I observed and maybe it is the result of the fact that handling rotations might not be quite clear. It goes like this:
* Load an object
[image]
* Rotate it around Z (45°)
[image]
* Rotate it around X (10°) 
[image]
* The object seems to be rotated around the **original** X axis - not the one connected to the object.
[image]
[image]

Moreover, if two consecutive rotations are applied to an instance of an object (with two instances), the first rotation applies to the selected instance only, but the second rotation applies to both instances.
If some scaling or resizing follows it is also applied to both instances, but translations are applied to the selected instance only.

bubnikv, 2019-03-01:
The instances may only differ in translation and rotation around Z. Their scale and rotation around Z is synchronized.

As of now, the scaling and rotation when entered from the side panel are in the original coordinate system of the object. We may change that in the future, or we may provide a switch between the World vs. Local coordinate system.

There is a bug in the handling of rotation field for the object. This bug will be fixed in the upcoming Slic3r 1.42.0-alpha7.

bubnikv, 2019-04-13:
Would you please test with beta1? The rotation hints shall be consistent now.

bubnikv, 2019-05-04:
no response, closing.

closed 2019-05-04 by bubnikv