prusa3d/PrusaSlicer #1781: Hostname incorrect While test is positive

caustinbakker, 2019-02-05:
Hello,

I am trying to use my hostname in the printer settings of slic3er PE. when i click test. it says it is working. When i go to send gcode to my printer it gives an error failed hostname error 6. When i place the ip of the server instead of the hostname it works perfectly. I have a dynamic ip and i would like to use the hostname but it is not working at the minute. Anyone know what the problem could be? 

### Version
_Version of Slic3r Prusa Edition used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-03-01:
For the name to work, as of now you need to install Bonjour resolver into your Windows box.

https://support.apple.com/kb/DL999?locale=en_US

Please let me know whether that worked.

Explanation: We resolve the Bonjour queries in Slic3r PE inside the "Browse" dialog only. We do not resolve the name to IP address using the Bonjour protocol if entered by name into the "Hostname, IP or URL" field. That will likely work if you install the driver by Apple, and we may implement the resolver directly into Slic3r one day.

bubnikv, 2019-04-13:
@vojtechkral may look into implementing a zeroconf DNS lookup on Windows without a system support

gusdleon, 2019-07-29:
@caustinbakker maybe you want to check my solution in #2689, **no** install needed, but you need to disable UPnP in your router

nospoon, 2020-03-07:
@gusdleon that's not a solution, you should not have to mess with your router settings like that.

gusdleon, 2020-03-07:
Sorry @nospoon I know it's no t a solution, but I discovered that always works when you do that, and stops working when you enable it

bacongobbler, 2020-07-21:
Disabling UPnP on the router did not resolve the issue for me. Installing Bonjour to resolve .local addresses did the trick though. Good ol' mDNS.

gusdleon, 2020-07-21:
well @bacongobbler for me... 
before the may update of windows (build 2004) windows stopped resolving .local addresses completely, and I refused to install Bonjour, but after the build 2004 magically windows started again resolving flawlessly local domains without installing anything additional.

addohm, 2021-08-06:
This is still an issue.  Testing the hostname works, sending a print job to the host does not.  Testing the IP works, sending a print job to the IP also works.  Need a solution for this that doesn't require me to ask people to disable router settings or install special and unrelated (on the surface) software just to be able to use a hostname to send a print job to the printer.

bubnikv, 2021-08-09:
I believe OctoPrint announces not only with Bonjour, but also with NetBIOS Service Discovery, which is natively supported by Windows. Thus if your network supports NetBIOS service discovery, sending a print job to a host should work. At least this is what we have tested. However we experienced an issue on some networks, which we hope we will fix in PrusaSlicer 2.4.0-alpha1.

https://en.wikipedia.org/wiki/Zero-configuration_networking

@kocikdav would you please chime in?

kocikdav, 2021-08-10:
Hello, there is a branch [dk_curl]( https://github.com/prusa3d/PrusaSlicer/tree/dk_curl) where we added option "Ignore certificate revocation checks". Would you @addohm please test if it helps? Here is a [build](https://www.dropbox.com/s/ilzgpq1593k0737/PrusaSlicer-dk_curl.rar?dl=0) of the branch, please note that this branch is far behind master right now, so other features might be missing or broken.

addohm, 2021-08-11:
> Hello, there is a branch [dk_curl](https://github.com/prusa3d/PrusaSlicer/tree/dk_curl) where we added option "Ignore certificate revocation checks". Would you @addohm please test if it helps? Here [...]

I will when I return to my office next week.

addohm, 2021-08-16:
> Hello, there is a branch [dk_curl](https://github.com/prusa3d/PrusaSlicer/tree/dk_curl) where we added option "Ignore certificate revocation checks". Also this branch should fix a zero DNS resolve issue when uploading a G-code from Slicer to OctoPrint.
> 
> OctoPrint shall respond to both Bonjour (Apple) and NETBIOS (Microsoft), thus one should not need to install Bonjour on Windows. However there was an issue when sending a G-code from PrusaSlicer, where two HTTP requests were sent to OctoPrint in succession, while NETBIOS zero DNS request was resolved for the first request only, not for the second. We don't know why the NETBIOS resolve fails for the 2nd HTTP request, however we worked around this issue by getting the resolved IP address from the 1st HTTP request and passing it to the 2nd HTTP request, which solved the issue at least in our tests.
> 
> Would you @addohm please test if it helps? Here is a [build](https://www.dropbox.com/s/ilzgpq1593k0737/PrusaSlicer-dk_curl.rar?dl=0) of the branch, please note that this branch is far behind master right now, so other features might be missing or broken.

Can you package this for me?  Really don't want to have to remove current vc++redist for old.  Thanks!

kocikdav, 2021-08-23:
@addohm Please try this [build](https://www.dropbox.com/s/k6xrpewn75alvt0/PrusaSlicer-2.4.0-alpha0%2B1152-dk-curl-new-win64-g59137d426-202108231536.zip?dl=0). Sorry for the delay.

bubnikv, 2021-09-23:
We believe the issue has been fixed in PrusaSlicer 2.4.0-alpha1.

closed 2021-09-23 by bubnikv

Photogad, 2022-02-12:
> We believe the issue has been fixed in PrusaSlicer 2.4.0-alpha1.

no

bubnikv, 2022-02-14:
@Photogad

I quite often have an issue with no-DNS resolve on Windows, however if it does not work in PrusaSlicer, it does not work in Windows either and I am not sure whether the issue is in Windows no-DNS resolver, my network or the Raspberry PI.

Does the discovery work in console for you?

Photogad, 2022-02-14:
> @Photogad
> 
> I quite often have an issue with no-DNS resolve on Windows, however if it does not work in PrusaSlicer, it does not work in Windows either and I am not sure whether the issue is in Windows no-DNS resolver, my network or the Raspberry PI.
> 
> Does the discovery work in console for you?

It works weird and makes no sense.  When I click "test" in printer settings of PrusaSlicer, it says everything is good. But when I slice and click upload and print, it says printer or host can not be found or something (i dont remember exact message).

Then I type in octopi.local to my web browser, and it loads into octopi UI,  and a few minutes after that I am able to upload and print from PS.

It's like loading octopi UI in browser wakes up the octopi to allow me to upload files to it from PS or something.

Photogad, 2022-02-14:
> The test functionality is implemented by @vojtechkral by sending and receiving UDP packets directly, while when connecting, we rely on Windows implementation of no-DNS resolve.
> 
> We may end up implementing our own UDP low level no-DNS resolve.
> 
> @Photogad What Windows version?

Windows 10 pro 64bit.

I did install bonjour service but it didn't fix the issue. Only fix is to load octoprint UI into browser to allow me to wake it up and allow PS to upload to it.

I should mention my octopi is off when I'm not printing and I turn it on before when I print. I suppose if I left it on, maybe it would stay connected.

Maybe it's more of an octopi issue where 3rd party apps can't connect to it unless the UI is loaded at least once per boot cycle.

vojtechkral, 2022-02-14:
I don't think Octopi get's "woken up", my guess is when the browser is used, it resolves the mDNS hostname and populates the global DNS cache, which then gets reused by PS. You can verify this by loading up OctoPi in the browser, then [flushing the DNS cache](https://grok.lsu.edu/article.aspx?articleid=13375), and then attempting upload in PS - if it fails, it was the cache.
Other than that I don't have much of a way to debug this, I have neither a windows box nor an octopi at the moment...

bubnikv, 2022-02-14:
I have an impression, that quite often an UDP no-DNS (mDNS) request or response gets lost, for example if the printer motors are running creating interference. We may implement our own mDNS resolver inside PrusaSlicer to be able to repeat, we may try to call the Windows DNS service multiple times. I don't know what will work.

vojtechkral, 2022-02-15:
BTW if I remember correctly, the custom DNS implementation is only for octopi discovery / local network scanning. The connection test (I _think_) goes through libcurl just like the upload. Which makes the problem pretty strange, I'm really not sure why it would work for one request and not the other...

vojtechkral, 2022-02-15:
@bubnikv Huh. Maybe [`CURLOPT_MAXCONNECTS`](https://curl.se/libcurl/c/CURLOPT_MAXCONNECTS.html) should be set to `1`? (I'm kind of just guessing here...)