prusa3d/PrusaSlicer #1783: 1.42.0-alpha5 "empty space" in preview view type dropdown menu

jhoblitt, 2019-02-05:
### Version

`Slic3rPE-1.42.0-alpha5+linux64-full-201902041815.AppImage`

### Operating system type + version

```
$ cat /etc/redhat-release 
Fedora release 29 (Twenty Nine)
```

### Behavior

* I believe this has long been present in slic3r.  I won't consider it a hard bug but it is a bit awkward...

[image]

bubnikv, 2019-02-05:
I am afraid this is the default behavior of the GTK widget library.

jhoblitt, 2019-02-05:
Ugh -- I guess it is doing that because the dropdown is on the bottom edge of the screen?  I can get the same thing to happen with gnome settings menus if I drag them to be just above the bottom of the screen. Double ugh.

jhoblitt, 2019-02-05:
And the bug tracker linked to from https://www.gtk.org/development.php is giving me an http 503... https://gitlab.gnome.org/GNOME/gtk/issues

bubnikv, 2019-02-05:
I am no more an active Linux user myself, and I was quite surprised by that
behavior. Why? I don't know.

jhoblitt, 2019-02-05:
I'm surprised I haven't noticed it before.  I don't do UI or gtk stuff but it doesn't look like there is a way to change the listing behavior either: https://developer.gnome.org/gtk3/stable/GtkMenu.html

I will attempt to file an issue on gtk3 if the bug tracker comes back up... should we close this issue or leave it open to prevent dups?

bubnikv, 2019-02-05:
close, please. We have the bug tracker overflowing already.

closed 2019-02-05 by jhoblitt