prusa3d/PrusaSlicer #1784: Slicing inconsistencies between 1.41.2 and 1.42.0 Alpha

cosmith71, 2019-02-05:
### Version
1.42.0 Alpha 5 (also present in other 1.42.0 versions)

### Operating system type + version
Windows 10 Home

### Behavior
This file is sliced differently (and looks worse) by 1.42.0 vs 1.41.2.  The file is admittedly challenging, but it used to work fine.  Other objects in this design show similar results.

File:  

https://www.thingiverse.com/download:5888456

1.41.2

[image]

1.42.0 Alpha 5

[image]

#### Project File (.3MF) where problem occurs

[Hub_90mm_v1.0.3mf.zip](https://github.com/prusa3d/Slic3r/files/2834061/Hub_90mm_v1.0.3mf.zip)

cosmith71, 2019-02-06:
EDIT:  This is an older version of the stl than is currently posted.  The new version doesn't have these problems, but the slicing difference on the old version is still noteworthy.

bubnikv, 2019-03-01:
There was a change introduced into Slic3r 1.42.0 series to fix issues #520 #820 #1029 #1364.
It was found out, that there are many STLs out there with thin cracks, gaps and slits, that the Slic3r 1.41.3 and older closed without asking. In the upcoming Slic3r 1.42.0-alpha5, we reverted to the old behavior with the default closing radius of 0.049mm, while providing a new parameter "slice_closing_radius" to satisfy the needs of #520 #820 #1029 #1364.

Closing.

closed 2019-03-01 by bubnikv