prusa3d/PrusaSlicer #1785: [Master commit 7719173] Sidebar is cut off

nebbywan, 2019-02-06:
### Version
Master commit 7719173

### Operating system type + version
Windows 10 Home version 1809 build 17763.292

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Printer/Filament/Profile/etc sidebar is cut off. Problem persists despite resizing the window or system display scaling. I'm running on a 4K notebook monitor and suspect this is related to the new HiDPI commits.
[image]
[image]

nebbywan, 2019-02-09:
Went back in time and confirmed the behavior started with commit 5207d31

bubnikv, 2019-02-09:
the HiDPI support on windows and linux is work in progress.

iamgnat, 2019-02-14:
Just a note to say I'm seeing the same thing with the Linux alpha5 AppImage version.

Similar issue with the `Wipe Tower Purge Volume Adjustment` dialog where the text box at the top of the window is cut off and not re-sizable.

YuSanka, 2019-02-14:
@iamgnat, thanks for your observations. Scaling of the Ramming/Wiping dialogs is a fixed in development branch (https://github.com/prusa3d/Slic3r/commit/8a31f9d658f3ded6afb45183d026ae32517540d1) and will be a part of a upcoming release.

nebbywan, 2019-02-15:
@YuSanka That build is looking great on my machine! Thanks for the fix!

nebbywan, 2019-02-21:
Fixed as of commit 478032a. Thank you!

closed 2019-02-21 by nebbywan