prusa3d/PrusaSlicer #1787: mmu2 with sequential printing doesn't change filament

tedder, 2019-02-06:
### Version
1.42.0-alpha5

### Operating system type + version
ubuntu 18.04 + gnome

### 3D printer brand / version + firmware version (if known)
mk3+mmu2

### Behavior
I'm keeping this short to see if it can be reproduced other than by me- I'm surprised I couldn't find another ticket that was similar.

Anyhow, to reproduce:
1. place two objects on bed
2. set sequential output
3. change one item to use a different extruder

When I hit 'slice now' (or watch it slice), it places one object with the correct extruder, then places the second object with *its* correct extruder, then changes the first object to use the second object's extruder. I'm showing two screenshots of the plate and sliced plate.

[image]
[image]

I can't figure out what happens that triggers this. Sometimes it always happens, now that I'm writing this ticket I can't get it to consistently reproduce. argh.

#### Project File (.3MF) where problem occurs
can't consistently reproduce.

tedder, 2019-02-06:
actually maybe I can repro, the problem is that I started using a super-simple shape. with at least one complex object perhaps it is happening? The parts in this 3MF come from the [dintid belted extruder](https://www.thingiverse.com/thing:2527297).

[slic3r_1787_from_bev4.zip](https://github.com/prusa3d/Slic3r/files/2834832/slic3r_1787_from_bev4.zip)

bubnikv, 2019-04-13:
We do not support sequential print with the MMU. 

How should the filament switch-over be performed? Should it run over a single layer wipe tower?

doliver-pulsemining, 2019-05-16:
> How should the filament switch-over be performed? Should it run over a single layer wipe tower?

 A single layer wipe tower may crash into parts when heading back to the same place on the bed multiple times. Ideally it would nice to print a skirt around each object equal (or greater) to the volume of the unload + load purge.

kylekyle, 2022-03-22:
This would be amazing.

closed 2026-09-29 by prusaslicer-bot[bot]