prusa3d/PrusaSlicer #1788: Linux: Plater right-click (context) menu opens only some of the time

progman32, 2019-02-06:
### Version
1.42.0-alpha5

### Operating system type + version

Both my Kubuntu 18.04.1 LTS machines suffer this problem.

One uses the nVidia proprietary driver. The other uses the default Intel GPU drivers.

### 3D printer brand / version + firmware version (if known)
i3 mk3 and i3 mk2

### Behavior

I apologize for the inconsistent repro, but this happens only some of the time (but frequent - at least 50% of right clicks are buggy for me).

1. Add a few objects to the plater.
2. Right click an object.

Expect: Context menu appears (`Reload from disk`, `Delete`, etc).
Actual: One of 3 things happen:

1. It works fine.
2. It selects the object but the context menu never appears.
3. Nothing happens at all.

It seems to happen more often if I don't have anything selected before doing the right click, but that's not the only condition in which it occurs. Happens regularly for various scenarios, like right-clicking the selected object.

enricoturri1966, 2019-02-06:
Working on it -> http://dev.prusa3d.com/browse/SPE-814

closed 2019-02-18 by enricoturri1966

progman32, 2019-03-03:
1.42-alpha7 confirmed to work on my end! Thanks!

Also btw for future readers - you need to right-click and HOLD to keep the menu open. Just tapping the right mouse button will open and immediately close the context menu. Don't confuse that for this bug :)

commit 0d25118e (2019-02-06): 1st attempt to fix #1788
commit efdb919e (2019-02-18): 2nd attempt to fix #1788
commit 022b573b (2019-02-18): 1st attempt to fix #1788
commit 362f6a06 (2019-02-18): 2nd attempt to fix #1788