prusa3d/PrusaSlicer #1789: Linux: Plater button clicks sometimes ignored

progman32, 2019-02-06:
### Version
1.42.0-alpha5

### Operating system type + version

Both my Kubuntu 18.04.1 LTS machines suffer this problem.

One uses the nVidia proprietary driver. The other uses the default Intel GPU drivers.

### 3D printer brand / version + firmware version (if known)
i3 mk3 and i3 mk2

### Behavior

This may be related to #1788 

I apologize for the inconsistent repro, but this happens only some of the time (but frequent - at least 50% of right clicks are buggy for me).

1. Go to Plater, optionally add objects.
2. Click on a button in the 3d view (Add object, arrange, lay on face, split, etc).

Expect: Appropriate action is invoked.
Actual: Usually the action is invoked, but sometimes nothing happens at all. I have to move the mouse and click again to make it work.

Interesting to note that the mouseover effect (bits of the icon turn orange) works 100% of the time.

bubnikv, 2019-02-06:
The window does not get the "Mouse entered" message after a dialog is
closed. That's why the 3D view does not get a keyboard focus. If you just
move the mouse outside the view and back inside, it will start working.
This behavior is specific to Linux/GTK. We shall certainly fix it somehow.

bubnikv, 2019-03-01:
This bug was different from #1788.

It is actually a bug in wxWidgets, where the window does not get a keyboard focus back after a dialog is closed. If you ask wxWidgets who has the focus, it tells "the 3D scene has it", but the keyboard is ignored.

In the upcoming Slic3r 1.42.0-alpha7, we reset the keyboard focus in "On window activate" event.

Closing.

closed 2019-03-01 by bubnikv

progman32, 2019-03-03:
1.42-alpha7 confirmed to work on my end! Thanks!