prusa3d/PrusaSlicer #1790: Alpha5 - infill not printing

thsturgill, 2019-02-06:
### Version
Slicer PE-1.42.0-alpha5+Linux64

### Operating system type + version
Fedora Linux version 27

### 3D printer brand / version + firmware version (if known)
i3 MK2.5 FW 3.5.1

### Behavior
When slicing preview shows infill - looks normal
When printing no infill was printed
[image]

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_
file:///home/thom/Desktop/peg-game-box.3mf

FidelCapo, 2019-02-08:
3mf file was not uploaded. Can you please try it again? Thank you

thsturgill, 2019-02-08:
[peg-game-box.3mf.zip](https://github.com/prusa3d/Slic3r/files/2845010/peg-game-box.3mf.zip)
The software would not allow 3mf format so I zipped it.

thsturgill, 2019-02-10:
I could not upload the 3mf file directly, so I zipped it and it allowed that to be uploaded.

bubnikv, 2019-03-01:
Sorry, I don't understand your issue. What infill is missing?

bubnikv, 2019-04-13:
no response, closing

closed 2019-04-13 by bubnikv