prusa3d/PrusaSlicer #1793: Can’t see build plate in preview.

Jabbawingo, 2019-02-07:
Version 1.42

FidelCapo, 2019-02-07:
I'm sorry, but this is too little information. What OS are you using? You can attach a screenshot and a Config file (File - Export - Export Config)? Thank you

Jabbawingo, 2019-02-07:
sorry.  Win10 [image]

`# generated by Slic3r Prusa Edition 1.42.0-alpha5+win64 on 2019-02-07 at 22:54:55
avoid_crossing_perimeters = 0
bed_shape = 0x0,200x0,200x200,0x200
bed_size_x = 68
bed_size_y = 120
bed_temperature = 50
before_layer_gcode = 
between_objects_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_angle = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 100
brim_width = 0
clip_multipart_objects = 0
colorprint_heights = 
complete_objects = 0
cooling = 0
cooling_tube_length = 5
cooling_tube_retraction = 91.5
default_acceleration = 0
default_filament_profile = ""
default_print_profile = 
deretract_speed = 0
disable_fan_first_layers = 3
dont_support_bridges = 1
duplicate_distance = 6
elefant_foot_compensation = 0
end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
ensure_vertical_shell_thickness = 1
exp_time = 8
exp_time_first = 35
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 60%
external_perimeters_first = 0
extra_loading_move = -2
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_colour = ""
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 1
fan_below_layer_time = 60
filament_colour = #29B2B2
filament_cooling_final_speed = 3.4
filament_cooling_initial_speed = 2.2
filament_cooling_moves = 4
filament_cost = 0
filament_density = 0
filament_diameter = 1.75
filament_load_time = 0
filament_loading_speed = 28
filament_loading_speed_start = 3
filament_max_volumetric_speed = 0
filament_minimal_purge_on_wipe_tower = 15
filament_notes = ""
filament_ramming_parameters = "120 100 6.6 6.8 7.2 7.6 7.9 8.2 8.7 9.4 9.9 10.0| 0.05 6.6 0.45 6.8 0.95 7.8 1.45 8.3 1.95 9.7 2.45 10 2.95 7.6 3.45 7.6 3.95 7.6 4.45 7.6 4.95 7.6"
filament_settings_id = "My Settings"
filament_soluble = 0
filament_toolchange_delay = 0
filament_type = PLA
filament_unload_time = 0
[... 133 lines cut ...]
thin_walls = 1
threads = 4
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 30
top_solid_layers = 3
travel_speed = 120
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
variable_layer_height = 1
wipe = 0
wipe_into_infill = 0
wipe_into_objects = 0
wipe_tower = 0
wipe_tower_bridging = 10
wipe_tower_rotation_angle = 0
wipe_tower_width = 60
wipe_tower_x = 180
wipe_tower_y = 140
wiping_volumes_extruders = 70,70
wiping_volumes_matrix = 0
xy_size_compensation = 0
z_offset = 0
`

FidelCapo, 2019-02-07:
We fixed similar issue, but I can not reproduce this one. Can you please zip and attach configuration file? You can find it by Menu - Help - Show Configuration File. Folder name is Slic3rPE-alpha. Thank you.

Jabbawingo, 2019-02-07:
[slic3r.zip](https://github.com/prusa3d/Slic3r/files/2840732/slic3r.zip)

FidelCapo, 2019-02-07:
Thank you. This issue should be same like https://github.com/prusa3d/Slic3r/issues/1772
Should be fixed in next release. Closing

closed 2019-02-07 by FidelCapo